protoloft/klipper_z_calibration #1: New update feedback

th3fallen, 2021-05-12:
1. document which config values are used far which moves, or add them to be overwritten in the zcalibrate config section
2. allow user to overwrite probe sample's for z calibration independently of normal probe samples.

Will add more items as i find them :p

TitusLabs, 2021-05-12:
Yes, they are already documented in short. But I'm already on this to make them all overwriteable...

TitusLabs, 2021-05-12:
Code and documentation are adapted now.

closed 2021-05-12 by TitusLabs

th3fallen, 2021-05-13:
@TitusLabs  Looks like the fast first probe and sample count isn't applied to the probe in the middle of the bed. Only the endstop and probe samples

TitusLabs, 2021-05-13:
hm, for me, it does.. you really activated this feature?

th3fallen, 2021-05-13:
yep this is my config atm

```[z_calibration]
switch_offset: 0.435  # using SSG-5H
max_deviation: 1.5
speed: 80             # the moving speed in x and y
samples: 2
probing_first_fast: true
probe_nozzle_x: 95   # coordinates for clicking the nozzle on the z-endstop
probe_nozzle_y: 300
probe_switch_x: 90   # coordinates for clicking the probe's switch on the z-endstop
probe_switch_y: 280
probe_bed_x: 150      # coordinates for probing on the print surface (the center point)
probe_bed_y: 150```

and the probe in the middle of the bed only samples 1 fast time

TitusLabs, 2021-05-14:
the samples property will not apply to the probing on the bed since I just call the probe there which is configured in the [probe] section. so, you mean only one sample? I think, you configured one sample on the probe ...but is it clicking twice on the bed?

th3fallen, 2021-05-14:
Ah ok. That makes sense. But the fast first probe shouldn't apply to that sample then if there's only 1 sample configured right?

TitusLabs, 2021-05-14:
yes, the fast first probe is not applied to any sample list at all!