protoloft/klipper_z_calibration #2: Self updating script

th3fallen, 2021-05-29:
No idea if you want to implement this but i added this to my moonraker.conf to add tracking of new versions of this lib

```[update_manager client z-calibrate]
type: git_repo
origin: https://github.com/protoloft/klipper_z_calibration.git
primary_branch: main
path: /home/pi/klipper_z_calibration```

then ssh'd in and did 
`ln -s /home/pi/klipper_z_calibration/z_calibration.py /home/pi/klipper/klippy/extras/z_calibration.py`

the only thing its lacking is a script to bounce the klipper service when it updates.

thought i'd share

TitusLabs, 2021-07-15:
Sorry, this does not work at all as I thought it could do. Currently I don't have time for this...

th3fallen, 2021-07-15:
Sorry I actually forgot to close this

closed 2021-07-15 by th3fallen

TitusLabs, 2021-07-15:
It is still an issue...