protoloft/klipper_z_calibration #5: error with multi mcu homing

wiyoon, 2021-07-09:
An error occurs when used with multi mcu homing branch.

pi@mainsailos:~/klipper $ git branch -a
* (HEAD detached at origin/work-homing-20210217)
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master
  remotes/origin/work-flash-20210519
  remotes/origin/work-fopdt-20180405
  remotes/origin/work-homing-20210217
  remotes/origin/work-linux-irq-20200607
  remotes/origin/work-mechaduino-20181205
  remotes/origin/work-pins-20201222
  remotes/origin/work-python3-20200612
  remotes/origin/work-tuning-20200805

---------------------------------------------------------------------------------
klippy.log
---------------------------------------------------------------------------------
Internal error on command:"G28"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 225, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 366, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 200, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/homing_override.py", line 60, in cmd_G28
    self.template.run_gcode_from_command(context)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 197, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/homing_override.py", line 23, in cmd_G28
    self.prev_G28(gcmd)
  File "/home/pi/klipper/klippy/extras/homing.py", line 252, in cmd_G28
    kin.home(homing_state)
  File "/home/pi/klipper/klippy/kinematics/corexy.py", line 65, in home
    homing_state.home_rails([rail], forcepos, homepos)
  File "/home/pi/klipper/klippy/extras/homing.py", line 209, in home_rails
    self.printer.send_event("homing:home_rails_end", self, rails)
  File "/home/pi/klipper/klippy/klippy.py", line 248, in send_event
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/extras/z_calibration.py", line 109, in handle_home_rails_end
    kin_spos = homing_state.get_stepper_trigger_positions()
AttributeError: Homing instance has no attribute 'get_stepper_trigger_positions'
AttributeError: Homing instance has no attribute 'get_stepper_trigger_positions'
Transition to shutdown state: Internal error on command:"G28"
Dumping gcode input 0 blocks
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[229.0, 355.0, 1.4304753823060659, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=200.0
Reactor garbage collection: (268.687623414, 181.198356706, 0.0)
MCU 'turbocan0' shutdown: Command request
clocksync state: mcu_freq=48000000 last_clock=13625396812 clock_est=(244.279 12248463101 48002638.387) min_half_rtt=0.000489 min_rtt_time=220.778 time_avg=244.278(875.719) clock_avg=12248463101.361(42036827636.792) pred_variance=1465532.097 clock_adj=(-0.860 48002166.750)
Dumping serial stats: bytes_write=9636 bytes_read=31806 bytes_retransmit=0 bytes_invalid=0 send_seq=924 receive_seq=924 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0
Dumping send queue 100 messages

TitusLabs, 2021-07-15:
Oh, thanks for the info! I cannot test all the branches and provide even branches with the fixes.
In this special case, the get_trigger_position('z') could work instead. But the z_homing variable is only used for the logging in the end..

TitusLabs, 2021-08-09:
With 0.8.0, I replaced this problematic code with the configured position_endstop value. So, it should work with this branch now...

closed 2021-08-09 by TitusLabs