protoloft/klipper_z_calibration #13: Homing instance has no attribute 'get_stepper_trigger_positions' (breaking change in Klipper?)

NameOfTheDragon, 2021-08-19:
It looks like today's Klipper commit (65ed55a43be0b995fde2d620ea838352efa7afb8) might have a breaking change that is killing the z_calibrate logic. When I home all axes, it fails when homing the Z axis with:
[image]

In Klippy.log...
[image]

I rolled back my Klipper install to an earlier commit and the problem went away.

Vassssko, 2021-08-20:
https://github.com/KevinOConnor/klipper/issues/4600#issue-975358446

Vassssko, 2021-08-20:
> It looks like today's Klipper commit (65ed55a43be0b995fde2d620ea838352efa7afb8) might have a breaking change that is killing the z_calibrate logic. When I home all axes, it fails when homing the Z [...]

Found the problem were most likely because of outdated z_calibration.py issue. Please doublecheck you have updated z_calibration.py, mind the following: https://github.com/protoloft/klipper_z_calibration#moonraker-updater. Current z_calibration.py have following strings commented:
[image]

DoritosVodka, 2021-08-20:
I had the same issue. Found re-running the z_calib install again corrected the files/issue... Tied to this moonraker update.

https://github.com/protoloft/klipper_z_calibration#moonraker-updater

NameOfTheDragon, 2021-08-20:
@Vassssko @DoritosVodka You guys were not wrong. I had auto-updated enabled via Fluidd and Moonraker, but as you guessed it looks like it was not updating the file correctly in the Klipper directory. I have now run the install script and it has generated a soft link. I have been able to update Klipper to the offending release and it's now all working again. I guess this can be closed now?

It would be really handy if it were possible to generate some sort of message during the auto-update process, like "Manual steps required" or something. I don't know the workings of the update process at all so don't know if that's even possible, but might be worth a look to save others from the bear trap that I fell into.

closed 2021-08-20 by NameOfTheDragon