protoloft/klipper_z_calibration #16: Probe samples exceed tolerance - Heaters remain on when error is raised

Marct27, 2021-09-01:
Hi,

Not sure if this is part of your macro or script or something that needs to be raised on the Klipper git.
Occasionally I get Probe samples exceed tolerance when calibrating the z offset and the print halts, but the heaters remain on.

Does your script use the standard probe commands from Klipper or should the script issue a shutdown if it errors out

Extract from the klippy.log
```
probe at 203.500,300.000 is z=0.753869
Probe samples exceed tolerance
Probe samples exceed tolerance
Probe samples exceed tolerance
Exiting SD card print (position 24234)
Stats 391442.8: gcodein=0 mcu: mcu_awake=0.002 mcu_task_avg=0.000007 mcu_task_stddev=0.000007 bytes_write=101248961 bytes_read=18515003 bytes_retransmit=9 bytes_invalid=0 send_seq=1853164 receive_seq=1853164 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119996802 z: mcu_awake=0.012 mcu_task_avg=0.000011 mcu_task_stddev=0.000014 bytes_write=19438025 bytes_read=18749160 bytes_retransmit=9 bytes_invalid=0 send_seq=670802 receive_seq=670802 retransmit_seq=7 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=28 stalled_bytes=0 freq=119997356 adj=120000505 rpi: mcu_awake=0.001 mcu_task_avg=0.000013 mcu_task_stddev=0.000017 bytes_write=583278 bytes_read=1797527 bytes_retransmit=0 bytes_invalid=0 send_seq=97189 receive_seq=97189 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999811 adj=50001130  heater_bed: target=110 temp=110.4 pwm=0.467 enclosure: temp=29.4 sysload=0.94 cputime=8162.670 memavail=758140 print_time=180684.271 buffer_time=0.313 print_stall=12 extruder: target=160 temp=160.0 pwm=0.035
Heater extruder within range of 160.000
Stats 391443.8: gcodein=0 mcu: mcu_awake=0.002 mcu_task_avg=0.000007 mcu_task_stddev=0.000007 bytes_write=101249051 bytes_read=18515199 bytes_retransmit=9 bytes_invalid=0 send_seq=1853171 receive_seq=1853171 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119996802 z: mcu_awake=0.012 mcu_task_avg=0.000011 mcu_task_stddev=0.000014 bytes_write=19438322 bytes_read=18749490 bytes_retransmit=9 bytes_invalid=0 send_seq=670816 receive_seq=670816 retransmit_seq=7 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119997356 adj=120000494 rpi: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000016 bytes_write=583284 bytes_read=1797556 bytes_retransmit=0 bytes_invalid=0 send_seq=97190 receive_seq=97190 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999808 adj=50001106  heater_bed: target=110 temp=110.4 pwm=0.467 enclosure: temp=29.4 sysload=0.94 cputime=8162.770 memavail=758108 print_time=180684.271 buffer_time=0.000 print_stall=12 extruder: target=160 temp=161.6 pwm=0.035
```

Many Thanks

Ramalama2, 2021-11-26:
Probe samples exceed tolerance, is natively klipper. And an error that klipper triggers. So it has actually nothing todo with the z calibration script.

The same error will happen if your probe tolerances will exceed on
quad_gantry_level
screw_tilt_adjust
bed_mesh_calibrate
Or everything else that exceed probe tolerance retryes.

So this should be actually a klipper issue.
Btw, the only workaround i know, is setting the idle timeout to 5 minutes or so and define in that idle timeout macro, that everything should be turned off.

As example:
```
[idle_timeout]
timeout: 300
gcode:
    {% if "z" in printer.toolhead.homed_axes %}
        G90
        G1 X175 Y350 Z50 F9000
    {% endif %}
    _CASELIGHT_OFF
    M84
    BED_MESH_CLEAR
    TURN_OFF_HEATERS
    SDCARD_RESET_FILE
    M107
```

This is not a proper fix, but at least anything.

Cheers ✌️

TitusLabs, 2022-02-18:
Ok, this is not so ideal. The error is raised by my plugin. And there is indeed a small difference in how Klipper this does. I will have a look if this has an impact on the heaters...

TitusLabs, 2022-02-20:
I did some testing and I will keep it as it is for now as it is the same behavior with other raised errors in Klipper like the same one in QGL.

closed 2022-02-20 by TitusLabs