protoloft/klipper_z_calibration #17: [FR] Support for BLTouch

hawkeyexp, 2021-09-08:
Hi, i tried your extra in knowledge of not tested with bltouch - all seems to work well but bltouch won't release pin to touch surface of endstop. Would it be possible to support the bltouch?

hawkeyexp, 2021-11-01:
It would be great if a support is possible - i know the voron guys does net really like the BLTouch by thermal reasons, but my voron 2.4 is running since july 2021 without any thermal caused problems :-)

electroleon, 2021-12-07:
I am not really sure how the BL Touch will work for this.
1. The pin is not really rigid when retracted so nozzle probing will not be accurate
2. The pin retracts higher than the nozzle, so you will effectively have a nozzle crash happen when you try to probe the BLTouch itself. So you would need a massive clearance around the nozzle probe to be able to have it work properly

hawkeyexp, 2021-12-11:
In general it should work with a bltouch but the offset for the z-endstop needs to be replaced with the bltouch one an the calculations

TitusLabs, 2022-02-18:
No, I'm sorry! A BLTouch would not work. It is impossible to measure the trigger point of the BLTouch to get everything in relations. For some use cases it might be sufficient to use a fixed offset for the BLTouch - but this would not be generally valid or even rock solid.

The BLTouch is a really nice idea to probe the bed - but only for open printers. In closed printers you get in trouble with higher temperatures. And this is not only because of the plastic pin on a hot print bed. There is also a metal clone out there with an IR trigger. But this would be much worse. You have the thermal expansion of the metal pin and changing trigger speeds of the electronics while getting hotter. This is why the simple dump switch is so unbeatable - believe it or not!!

TitusLabs, 2022-02-22:
Ok, I thought about it again. So, it might be possible... 

The difference in such a system is, that the nozzle is lower as the BLTouch. 
- So it needs an abs function in the equation.
- And it must be possible to drive the BLTouch on the endstop not crashing the nozzle into the bed.
- Then it just needs the switch_offset of the BLTouch.
 
But don't know how this would perform... And I cannot give any support on this. Is there someone who wants to try it?

hawkeyexp, 2022-03-04:
I have solved the problem myselve and created and simple but perfect working solution which is now running since weeks on more than my single printer.

https://github.com/hawkeyexp/auto_offset_z

For calculation it is "easy" - probing endstop surface and bed surface and the diff is the general offset - additional the way the endstop switch needs for moving until triggering the switch (datasheet ~ 0.5mm) - for the last adjust until the switch really triggers i added a manual offset in the config section. On first print the nozzle is a bit higher than needed which can be adjusted by babystepping and this value which was needed must be written to the config as manual adjust - that's it!

TitusLabs, 2022-03-26:
Uh, I like it! So, you basically do it the other way around and define a switch_offset for the z endstop :open_mouth: 
I will have a look how I could integrate such a use case..

LeandroScovino, 2022-05-23:
This would be amazing. I have a previously closed source firmware 3D printer from Sethi 3D here in Brazil. This has a piezo button switch outside the bed, comparable to Voron's Z Endstop and a IR/Mechanical touch (like BLTouch) on the extruder. 

The original firmware first home on the piezo button, than hit the nozzle on the piezo, calculating the difference offset disregarding the bed. 

A year ago, I've retrofitted this printer with a new MCU to use klipper, as it has 4 Z motors... and this week I started working on writing a plugin to calculate the offset. I believe @hawkeyexp got this working.

Congrats you both!

[image]

TitusLabs, 2023-04-05:
The question is, what do you want to do with it? This is certainly a good method to measure the offset of the BLTouch. Go for it!

The BLTouch has its limits. If the printer is enclosed and the print bed is over 100 degrees and the X-axis is a good 90 degrees, then a lot changes. The resistances and capacitances in the electronics change... and the offset shifts.

For most printers the BLTouch is great! The offset should not change much once it is calibrated. Only if you change the nozzle, you have to repeat the offset procedure...

I don't see something I can do here with this plugin regarding the BLTouch :thinking:

ameswarb, 2024-10-03:
I was able to get this to work with my CR Touch with [these modifications](https://github.com/protoloft/klipper_z_calibration/compare/master...ameswarb:klipper_z_calibration:use_bltouch).

I am experiencing an issue however, and I'm not sure if it's caused by my changes.

With the script enabled, at around ~2 - 2.35mm (10 - 12 layers) of print height it stops properly lifting the Z-axis, causing excessive blobbing / zits for the next ~4.5 - 8mm. My 30mm calibration cubes where this issue is present are ~26.44mm tall.  

Here's my potentially relevant blocks of printer config.

```
[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
position_min: -3
position_max: 240
endstop_pin: PC2
position_endstop: 4.996

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 999999

[z_calibration]
nozzle_xy_position: 253, 117.5
switch_xy_position: 253,117.5
bed_xy_position: 117.5,117.5
switch_offset: 6.186
speed: 50.0
offset_margins: -6.0,6.0

[bltouch]
sensor_pin: ^PC14
control_pin: PA1
x_offset: -31.8
y_offset: -40.5
pin_up_reports_not_triggered: false

[bed_mesh]
speed: 120
horizontal_move_z: 15
mesh_min: 30, 20
mesh_max: 200, 195
probe_count: 9, 9
mesh_pps: 2, 3
algorithm: bicubic
bicubic_tension: 0.2
fade_start: 1
fade_end: 10
fade_target: 0
```

[image]

SSpecken, 2024-12-27:
@ameswarb - I've tried your patch to get z_calibration to work with my BLTouch, but it doesn't: probing z-endstop is fine, but after the nozzle of the BLTouch doesn't raise down so nothing is measured. Do you have got any other modifications you didn't mention probably?

TitusLabs, 2026-09-03:
Closing after four years: BLTouch-style probes are not supported, and 2.0.0
states that in the README. The trigger point can't be measured against the
endstop with the repeatability the calculation needs, especially in an enclosed
chamber.

For BLTouch/CR Touch setups, @hawkeyexp's https://github.com/hawkeyexp/auto_offset_z
takes the opposite approach and is the right tool. Leaving this open only
suggests it might still arrive — it won't.

closed 2026-09-03 by TitusLabs