protoloft/klipper_z_calibration #18: Edit Description in the z_calibration.cfg

xenon2008, 2021-09-19:
Hi there,

I'm still fairly new to the Klippper and Voron world and in the last few days I tried to install or even set up the klicky with Auto Z calibration.

However, I was always too close to the bed until I noticed that my offset, which was 0.42 or 0.08, for example, was not taken into account!

Here I became skeptical and realized my mistake ..
I triggered the button of the Omron switch on the Z endstop and not the housing of the switch.

The reason for this was the assumption because of the text here ...
[image]

Could that be changed so that other users don't fall for it and the switch housing would will be used?

Kind Regards from Upper Austria

arkeet, 2022-01-24:
I have opened PR #33 for this.

TitusLabs, 2022-02-18:
Thanks! I have already merged it...

This is really important! And there are people doing this by purpose - I do not support this!

closed 2022-02-18 by TitusLabs