protoloft/klipper_z_calibration #19: Add "Safety Probe check" to Calibrate_Z macro ?

DoritosVodka, 2021-10-01:
I just had an issue during calibrate_z use and wonder if a probe "safety check" can be added to the macro...

Running the Klicky Probe and issued a calibrate_z command/macro.

- It attached the probe successfully like normal, then moved to the Z endstop switch position.
- During the move the klicky probe got knocked off by the cleaning brush (did not lift high enough, my fault and fixed).
- It did not "sense" the probe was knocked off and probed the nozzle on the z endstop switch. Which was fine.
- Then tried to probe the switch body on the z endstop switch. Since the probe was knocked off it kept driving down until the nozzle buried itself in the bed.

The attach_probe and dock_probe has the checks built-in. But, if the probe gets knocked off during movement the calibrate_z function does not seem to check between it's steps for this. Is it possible to add a query_probe for an "open" state after movements to make sure it's still attached ?

Frix-x, 2021-10-04:
Not sure if this belongs to the calibrate_Z plugin as there is a lot of type of probes and the way they work or are wired will change the behavior.
However, you can use the check_probe macro from klicky just before doing a calibrate_z and it will work the same.

DoritosVodka, 2021-11-07:
It looks like klicky_probe git was updated and has a constant check in it now to make sure the probe is still connected.

closed 2021-11-07 by DoritosVodka