protoloft/klipper_z_calibration #20: [FR] Be able to configure the plugin at the runtime

Frix-x, 2021-10-04:
Hi,

This is just a simple feature request to this plugin : I would love to be able to specify custom parameters at the call of the CALIBRATE_Z macro be able to do:
`CALIBRATE_Z PROBE_BED_X={computed_value} PROBE_BED_Y={computed_value} [OTHER_PARAMETERS=...]`

Use case : I'm doing a specific custom bed_mesh macro at every print start that is adapting the probing to the first layer size from the slicer. This system make the bed_mesh faster while still allowing for a very precise mesh: like 9x9 on the full plate but maybe 3x3 for a small part alone in the center or even around a small part in a corner if wanted. That mean the relative_reference_index change every time.
In fact today it's already working good for centered parts, but recently I got the center of my PEI sheet damaged and started to print off-centered. My custom bed mesh is already following the parts on the bed but the calibrate_z is still probing the center => This means that I got a deviation from the relative_reference_index that is not at the same position.

What do you think about this ?
I'll try to do a PR if I find time to do so :)

julianschill, 2021-11-25:
I also wanted to only mesh the print area.
I made a PR for taking the RRI from the current mesh: https://github.com/protoloft/klipper_z_calibration/pull/23

Frix-x, 2021-11-26:
Thanks for that. It's a good starting point !

But for my use case it's not enough as I'm doing the mesh at the very end of the ```PRINT_START``` macro, and after the ```CALIBRATE_Z```. I do it like that to mesh just before the print at the very last moment and also because I'm not sure how the offset would be calculated from the Z if you calibrate the Z after the mesh...

At the moment, what I'm using is a small modification of the plugin made by a user of the Voron Discord to quiclky add a new method:
```SET_Z_CALIBRATION_POSITIONS [PROBE_NOZZLE_X=<x_position>] [PROBE_NOZZLE_Y=<y_position>] [PROBE_SWITCH_X=<x_position>] [PROBE_SWITCH_Y=<y_position>] [PROBE_BED_X=<x_position>] [PROBE_BED_Y=<y_position>]```
It's working very well and an easy modification to the plugin. However, I think it's not the cleanest way to do it and that's why I did not pushed it here today.

BTW, if you want to have a look at my mesh macro, here it is : https://github.com/Frix-x/klipper-voron-V2/blob/main/macros/probing/bed_mesh.cfg

julianschill, 2021-11-26:
As the bed mesh is relative to the RRI, the Z offset doesn't matter at all for the bed mesh. So it is safe to do the calibration after the mesh at the RRI. The advantage is, that the position of the RRI is only calculated once in the bed_mesh code.

I am wondering if this all could be integrated, because we already know the height of the RRI from the bed_mesh and the height of nozzle on endstop from homing (should be exactly position_endstop of Z). So the only thing you would need is the switch on endstop height. So theoretically you could do an all in one: Home, get Switch on Endstop, run mesh. Then pick the values and calculate the Z Offset.

Ramalama2, 2021-11-26:
Hey guys, i would suggest that you try out this macro for making a bed mesh only in your print are:
https://github.com/Turge08/print_area_bed_mesh

This macro reconfigures the relative reference index to the middle of the bed meshed area.
This means:
1. Calibrate_z uses by default, if you don't define probe_bed_x and probe_bed_y to:
```
probe_bed_x: default from relative_reference_index of bed_mesh
probe_bed_y: default from relative_reference_index of bed_mesh
```
2. After you made your bed mesh, you do a calibrate_z afterwards.

I guess this is exactly, what you guys wanted. So probably this issue can be closed then.

Cheers ✌️

julianschill, 2021-11-26:
Z_calibration reads the probe position once on startup.

julianschill, 2021-11-26:
No this doesn’t solve it. This macro does nothing different than my macro. Z_calibration determines the position of the rri point once on startup and probes always there. My pull request changes that to determining the point again on every call of CALIBRATE_Z.

Ramalama2, 2021-11-26:
> No this doesn’t solve it. This macro does nothing different than my macro. Z_calibration determines the position of the rri point once on startup and probes always there. My pull request changes that to determining the point again on every call of CALIBRATE_Z.

Ooh, i thought it is dynamic already. Yeah then this won't be a solution, sorry.
Thanks for the PR

garethky, 2021-12-04:
+1 for passing the position to probe to the command. I'm also trying to bed mesh only a sub-area of the bed. When the probe point for Z_CALIBRATION is outside the mesh area I get bad z offsets. Works perfectly otherwise.

electroleon, 2021-12-07:
So I have absolutely no idea how Git works exactly. But you can modify this part in the code to have this work.

from Line 136 in the python file

   def cmd_CALIBRATE_Z(self, gcmd):
		probe_bed_x = gcmd.get_float("PROBE_BED_X", self.probe_bed_site[0], 0)
		probe_bed_y = gcmd.get_float("PROBE_BED_Y", self.probe_bed_site[1], 0)
		probesite = [probe_bed_x,probe_bed_y,None,]
		if self.state is not None:
		
			raise self.printer.command_error("Already performing CALIBRATE_Z")
			return
		self._log_config()
		state = CalibrationState(self, gcmd)
		state.calibrate_z(probesite)

electroleon, 2021-12-07:
I have created a PR for this.

TitusLabs, 2022-02-20:
You really need to calibrate Z on the RRI point - nothing else. So, I don't like the idea to set the point by X and Y coordinates. And yes, if you change this RRI on the bed you need to calibrate Z again. But this should not be a problem at all..

So, from my point of view the PR #23 is all you need for this.

closed 2022-02-20 by TitusLabs