protoloft/klipper_z_calibration #21: CALIBRATE_Z command raised internal eeror

dukeduck1984, 2021-11-10:
Hi,

When I executed `CALIBRATE_Z`, an error was raised as following:
```
16:12:09  $ CALIBRATE_Z
16:12:09  // Klipper state: Shutdown
16:12:09  !! Internal error on command:"BASE_CALIBRATE_Z"
16:12:09  !! Internal error on command:"CALIBRATE_Z"
```
And also a pop warning message `float argument required, not NoneType`.

This is what I found by looking at the traceback in the `klippy.log`:
```
Internal error on command:"BASE_CALIBRATE_Z"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 120, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 120, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/z_calibration.py", line 140, in cmd_CALIBRATE_Z
    self._log_config()
  File "/home/pi/klipper/klippy/extras/z_calibration.py", line 233, in _log_config
    self.probe_bed_site[0], self.probe_bed_site[1]))
TypeError: float argument required, not NoneType
Transition to shutdown state: Internal error on command:"BASE_CALIBRATE_Z"
```

I commented out both `probe_bed_x` & `probe_bed_y` in `z_calibration.cfg`, but did have  `relative_reference_index` configured in `bed_mesh` section in `printer.cfg`.

Is there anything else I might have missed?

Thanks for the help in advance.

Kaiyuan

dukeduck1984, 2021-11-10:
The same error occured even when I had set the value for `probe_bed_x` & `probe_bed_y`.

dukeduck1984, 2021-11-10:
I had tried commenting out line 225 & 233 in method `_log_config()`, and the same error occured again but the traceback pointed to `probe_switch_site` - apparently the list `probe_switch_site` contained `None` as well.  So I commented line `self._log_config()` from method `cmd_CALIBRATE_Z()` and then it worked.

Any way to fix this?

porkyneal, 2021-11-23:
I'm too struggling with this setup on a clean install. I think the error is the same.

julianschill, 2021-11-25:
I figured the problem is, that some variables get initialized on homing of Z. So they are None until Z is homed. https://github.com/protoloft/klipper_z_calibration/blob/cb68379941dfb6189cc9b2a4904cc3d6ce23024f/z_calibration.py#L116-L131

Best thing would be to query if Z is homed before continuing. It doesn't make sense to calibrate without Z being homed.

TitusLabs, 2022-02-20:
I've added a check for homed axes.

closed 2022-02-20 by TitusLabs