protoloft/klipper_z_calibration #22: How to use it

laserbeak, 2021-11-19:
Hi, I don't understand what I'm supposed to be doing [here](https://github.com/protoloft/klipper_z_calibration#how-to-use-it)
Should I replace `BASE_CALIBRATE_Z `with `BED_MESH_CALIBRATE`?
My code looks like this so far:
```

[gcode_macro CALIBRATE_Z]
rename_existing: BED_MESH_CALIBRATE
gcode:
    G28
    M117 Z-Calibration..
   # _SET_LOWER_STEPPER_CURRENT  # I lower the stepper current for homing and probing 
    Attach_Probe                # a macro for fetching the probe first
    #BASE_CALIBRATE_Z
    Dock_Probe                # and parking it afterwards
    #_RESET_STEPPER_CURRENT      # resetting the stepper current
    M117
```

Ramalama2, 2021-11-26:
> Hi, I don't understand what I'm supposed to be doing [here](https://github.com/protoloft/klipper_z_calibration#how-to-use-it) Should I replace `BASE_CALIBRATE_Z` with `BED_MESH_CALIBRATE`? My code looks like this so far:
> 
> ```
> 
> [gcode_macro CALIBRATE_Z]
> rename_existing: BED_MESH_CALIBRATE
> gcode:
>     G28
>     M117 Z-Calibration..
>    # _SET_LOWER_STEPPER_CURRENT  # I lower the stepper current for homing and probing 
>     Attach_Probe                # a macro for fetching the probe first
>     BED_MESH_CALIBRATE
>     Dock_Probe                # and parking it afterwards
>     #_RESET_STEPPER_CURRENT      # resetting the stepper current
>     M117
> ```

You don't need to do anything there.

> Hi, I don't understand what I'm supposed to be doing [here](https://github.com/protoloft/klipper_z_calibration#how-to-use-it) Should I replace `BASE_CALIBRATE_Z` with `BED_MESH_CALIBRATE`? My code looks like this so far:
> 
> ```
> 
> [gcode_macro CALIBRATE_Z]
> rename_existing: BED_MESH_CALIBRATE
> gcode:
>     G28
>     M117 Z-Calibration..
>    # _SET_LOWER_STEPPER_CURRENT  # I lower the stepper current for homing and probing 
>     Attach_Probe                # a macro for fetching the probe first
>     BED_MESH_CALIBRATE
>     Dock_Probe                # and parking it afterwards
>     #_RESET_STEPPER_CURRENT      # resetting the stepper current
>     M117
> ```

You don't need anything there.

Citus made an example, for configurations where the magprobe doesn't attaches automatically.
So that section you link for is exactly for that, just an example how to manually modify the calibrate_z "macro".
Or to be more explicite, what to do before and after if you execute calibrate_z.

For your bed_mesh macro example, you may have or may not, attach the magprobe either. But this doesn't has anything todo with calibrate_z.

Calibrate_z and bed_mesh are separate things.
Cheers

laserbeak, 2021-11-27:
is there a walkthrough that teaches you how to use it?
I'm pretty confused :)

Ramalama2, 2021-11-27:
What happens if you simply execute calibrate_z?

laserbeak, 2021-12-01:
> What happens if you simply execute calibrate_z?
running calibrate_z will home x y and z, attach the probe then immediately dock it.
> We simply include it usually at the end of print_start...
What do you mean by this?

Ramalama2, 2021-12-01:
> > What happens if you simply execute calibrate_z?
> 
> running calibrate_z will home x y and z, attach the probe then immediately dock it.
> 
> > We simply include it usually at the end of print_start...
> 
> What do you mean by this?

Omg...
Yes because you use the macro above.... Comment out the macro or delete it then check what calibrate_z does 🤦

laserbeak, 2021-12-01:
...sorry but the english during this whole conversation hasn't been very clear....(but I'm not judging you because of that, so what's with the attitude problem? 🤦 )

Not surprisingly, commenting out the macro does exactly what I thought it would...
```
 16:28:20  $ calibrate_z
16:28:20  // Unknown command:"CALIBRATE_Z"
```

Ramalama2, 2021-12-01:
> ...sorry but the english during this whole conversation hasn't been very clear....(but I'm not judging you because of that, so what's with the attitude problem? 🤦 )
>
> Not surprisingly, commenting out [...]

Yes, because you didn't installed the plugin correctly 🤦

laserbeak, 2021-12-01:
OK, well shame on me then? :man_shrugging:

I didn't exactly find documentation to do this. it seems to be fine from fluidd...
is there any documentation to properly install a plugin? or is asking setting myself up for more shame?

Ramalama2, 2021-12-01:
Ssh into your pi with putty...
But with the "pi" user.
If you ssh with root user, you need to enter:
`su - pi`

Then:
```
cd /home/pi
git clone https://github.com/protoloft/klipper_z_calibration.git
cd klipper_z_calibration
chmod +x install.sh
./install.sh

```
Then you edit your moonraker.conf
located in:
```
cd /home/pi/klipper_config/
nano moonraker.conf
```
And you add this to it:
```
[update_manager client z_calibration]
type: git_repo
path: /home/pi/klipper_z_calibration
origin: https://github.com/protoloft/klipper_z_calibration.git
install_script: install.sh
```

Then you restart klipper, and you will get some errors.
Those means, that you have to add this to your config:
https://github.com/protoloft/klipper_z_calibration/blob/master/configuration/z_calibration.cfg

You can simply download it and include it with:
[include z_calibration.cfg]
In your printer.cfg.

Read what those Options means and set them correctly.

Cheers

laserbeak, 2021-12-02:
That's what I didn't do!! 🤦
I never included z_calibration.cfg in my printer.cfg 🤦🤦🤦
Thanks so much!!
I ran calibrate_z and it doesn't attach my probe though...

```
 03:02:23  $ CALIBRATE_Z
03:02:23  // Probe already docked!
03:02:23  // probe: TRIGGERED
03:02:28  !! No trigger on probe after full movement
03:02:28  !! No trigger on probe after full movement
```

I've manually attached the probe and ran calibrate_z again

 ```
03:07:11  $ CALIBRATE_Z
03:07:11  // Docking Probe
03:07:13  // probe: TRIGGERED
03:07:18  !! No trigger on probe after full movement
03:07:18  !! No trigger on probe after full movement
```
​
This isn't what's supposed to happen, right? 
:headstone: :sweat_smile:
​As you've suggested earlier, I put it at the end of my print_start? 
I was hoping i could test it before printing something.

Ramalama2, 2021-12-02:
> 

This is the minimal config, you basically need only to define this:
```
[z_calibration]
probe_nozzle_x: 271.5 #your endstop x pos
probe_nozzle_y: 353 #your endstop y pos
probe_switch_x: 265.7 #see the picture attached below
probe_switch_y: 331.9 #see the picture attached below
switch_offset: 0.50 #d2f, read the docs
max_deviation: 1.0 #1.0 is a good default value
speed: 50
probing_first_fast: true
start_gcode: ATTACH_PROBE #this is what you miss
end_gcode: DOCK_PROBE #this you miss surely too
```

[image]
probe_switch_x and y, are the coordinates of that point. Z-endstop needs to hit the "body" of the microswitch, not the button of the microswitch.

You can replace basically your config with my above, change the coordinates and your should be good 😘

Ramalama2, 2021-12-02:
No trigger on probe after full movement

This means simply that your klicky isn't working 😂

laserbeak, 2021-12-03:
> No trigger on probe after full movement
> 
> This means simply that your klicky isn't working 😂

Yes, it doesn't work when trying the calibrate_z macro. But I use it all of the time. This is what leads me to believe that my z_calibration.cfg isn't configured correctly.  :thinking:

> No contact or whatever. And according to your good picture from the side, there is Filament sticking out. You need to calibrate-z with turned on hotend and with a small retraction before, else it won't be accurate if the filament sticks out.
Damn it, I knew I should've removed that filament from my "awesome" ( you made a typo there ) picture. That's from changing filaments. I don't normally have that oozing from my nozzle.
In this particular case, does it really make a difference, since it's not even touching the endstop?

AndrewTheWookiee, 2021-12-03:
I was getting the exact same error as you, with the exact same behavior with the nozzle stopping just above the z endstop. Try changing your position_min in the z_calibration.cfg. It had a default of 0, and when I changed it to -1 everything started working as intended.

laserbeak, 2021-12-04:
Hi!
Mine was 0 also. I've set it to  my position_endstop value (-0.74) and it actually starts to do its thing, but I get an error when sampling from the z endstop with the probe body:

 ```
13:42:07  $ CALIBRATE_Z
13:42:07  // Attaching Probe
13:42:10  // probe: open
13:42:15  // probe at 225.000,350.000 is z=-0.730000
13:42:16  // probe at 225.000,350.000 is z=-0.727500
13:42:18  // probe at 225.000,350.000 is z=-0.725000
13:42:20  // probe at 225.000,350.000 is z=-0.725000
13:42:22  // probe at 221.000,330.000 is z=6.347500
13:42:24  // probe at 221.000,330.000 is z=6.355000
13:42:26  // probe at 221.000,330.000 is z=6.357500
13:42:27  // probe at 221.000,330.000 is z=6.360000
13:42:27  !! Move out of range: 12.000 -7.750 22.200 [0.000]
13:42:27  !! Move out of range: 12.000 -7.750 22.200 [0.000]
```

AndrewTheWookiee, 2021-12-04:
That looks like it successfully probed the nozzle (225,350), then probed the switch body (221,330), then it looks like it's trying to move to (12,-7.75)

What are your probe_bed_x and probe_bed_y values? They should be in the middle of your bed.

For reference, here's my console from a successful calibration:

`13:08:26 
$ calibrate_z
13:08:26 
// Attaching Probe
13:08:28 
// probe: open
13:08:33 
// probe at 266.000,390.000 is z=-0.767438
13:08:35 
// probe at 266.000,390.000 is z=-0.759950
13:08:37 
// probe at 266.000,390.000 is z=-0.757455
13:08:39 
// probe at 266.000,390.000 is z=-0.757455
13:08:44 
// probe at 261.002,370.000 is z=0.999641
13:08:46 
// probe at 261.002,370.000 is z=0.999641
13:08:48 
// probe at 261.002,370.000 is z=1.002136
13:08:50 
// probe at 261.002,370.000 is z=1.002136
13:08:57 
// probe at 188.001,164.003 is z=1.873196
13:08:58 
// probe at 188.001,164.003 is z=1.885676
13:08:59 
// probe at 188.001,164.003 is z=1.885676
13:09:00 
// probe at 188.001,164.003 is z=1.885676
13:09:00 
// Z-CALIBRATION: ENDSTOP=-0.730 NOZZLE=-0.757 SWITCH=1.002 PROBE=1.886 --> OFFSET=-0.423915
13:09:00 
// Docking Probe
13:09:05 
// probe: TRIGGERED`

laserbeak, 2021-12-05:
Ah..that's what it was. I misread the comments in the config file.
my x and y were 12 when they should've been 175. (390...wow, that's a big print bed).
Thanks!!!

laserbeak, 2021-12-05:
Closing now. Benchy looks great ❤️‍🔥 . 
Thanks again for the help!!

closed 2021-12-05 by laserbeak

AndrewTheWookiee, 2021-12-05:
Awesome :)

Glad everything worked out.