protoloft/klipper_z_calibration #24: Tool head crash when probe detaches

CptSandwich, 2021-11-29:
I've just encountered a tool head crash when the plugin was attempting to touch the probe to the Z endstop pin. One of the magnets broke free from the probe, which must have occurred after the attachment procedure for it to have passed the attachment check. However, shortly thereafter the probe has fallen from the tool head and continued with the Z calibration procedure resulting in the nozzle crashing into the bed, because no probe was present to touch the Z endstop pin. 

My assumption is that because klipper/the plugin is expecting to see a change in Z endstop state, it's not checking the probe state and is therefore unaware of the presence (or otherwise) of the probe. Since it must be attached to avoid a crash, is it possible to throw an error if the probe state is open during the Z calibration?

Tim2309, 2021-12-20:
I have the same problem, on first test my probe failed to attach.
i was able to stop the printer but a automatic stop in the script would be nice

iNDeX999, 2021-12-22:
Same Problem, lost the probe and the tool crashed in the bed.

eduncan911, 2021-12-25:
I've always wanted some sensorless limit like for when the nozzle goes past the endstop.

Tim2309, 2022-01-25:
The solution is really simple.
The original script contain the function already, all you have to do is to open the file "z_calibration.py"
and remove the # in the following lines:

#ERROR_NOT_ATTACHED = "Probe switch not closed - Probe not attached?"

 #time = self.helper.printer.lookup_object('toolhead').get_last_move_time()
        #probe = self.helper.printer.lookup_object('probe')
        #if probe.mcu_probe.query_endstop(time):
        #    raise self.helper.printer.command_error(ERROR_NOT_ATTACHED)
        #    return

now the printer recognise if the probe is missing while attaching it.
The script does NOT check a second time if the probe is still attached before lowering to the z-endstop

TitusLabs, 2022-02-18:
Yes, I'm not sure about this since I would limit this extension to a normally closed probe setup - but maybe this is not an issue?
And I even wanted this check to be separated from this extension...
But, loosing the probe in between seems to be a serious danger?

TitusLabs, 2022-02-20:
I've added this check before using the probe :thumbsup:

closed 2022-02-20 by TitusLabs