protoloft/klipper_z_calibration #27: CALIBRATE_Z - Docking probe before moving to press z-endstop.

keenangodburn, 2021-12-08:
This is for a Voron 2.4 with the jlas1/klicky-probe.

I've installed the probe and it's working great for QGL and Bed Mesh, but after installing klipper z calibration, when I run Calibrate_Z, the printhead attaches the probe, docks it, then moves over to the z endstop where the probe fails since there is no probe attached. 

I've attached my config files as .txt below.

[z_calibration.txt](https://github.com/protoloft/klipper_z_calibration/files/7672356/z_calibration.txt)
[printer.txt](https://github.com/protoloft/klipper_z_calibration/files/7672357/printer.txt)
[Klicky_Probe.txt](https://github.com/protoloft/klipper_z_calibration/files/7672358/Klicky_Probe.txt)

TitusLabs, 2022-02-19:
Is this still an issue? Maybe you just configure the `start_gcode` and `end_gcode` properties with your attach and detach Macro names?

TitusLabs, 2022-03-26:
Ok, I will close this now..

closed 2022-03-26 by TitusLabs