protoloft/klipper_z_calibration #28: prints a little too close to the bed.

laserbeak, 2021-12-08:
Hi, 
what do I need to adjust in order for kzc to print a little further from the bed?

bmkahl, 2021-12-08:
I am having the same issue. I have calibrated the limit switches, but it keeps adjusting the Z offset too low.

laserbeak, 2021-12-09:
I've made my endstop a little taller to be more level with the bed when the switch is activated, and edited `position_endstop` in printer.cfg and `position_min` in z_calibration.cfg, but it didn't do much, if anything at all.

bmkahl, 2021-12-09:
I was able to get the right offset by adding SET_GCODE_OFFSET Z_ADJUST=1 into my macro and then adjusting the switch_offset. Note that I do not have a Voron but rather have a custom built machine that has a limit switch on a SG90 servo. 

[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    OG28                  #only home if needed
    SET_SERVO SERVO=my_servo ANGLE=0                
    BASE_CALIBRATE_Z
    SET_SERVO SERVO=my_servo ANGLE=120
    SET_GCODE_OFFSET Z_ADJUST=1

laserbeak, 2021-12-11:
I'm glad it works, but doesn't that defeat the purpose of kzc?
how close was it to your bed that you had to use 1?

bmkahl, 2021-12-13:
It was about .65mm too low consistently even when the switch offset was set to 0.01.

laserbeak, 2021-12-13:
I started at .5, now I'm at .38. I'm printing things and testing ever now and then. I'll probably be down to .34 before the week's over.

eduncan911, 2021-12-26:
I'm experiencing the same issue with a "load cell" nozzle probe that touches the actual nozzle.

It has a flex of about 0.2mm, that I need to adjust for.  I previously was using `probe->z_offset` set to `-0.2` for perfection before adding this script.

Now, I cannot seem to set this offset in any code.  The previous `z_offset` doesn't seem to change regardless.  And using `z_calibration->switch_offset` only seems to change the Z offset being reported... 

**My problem is whenever I enter "0mm" on Z, it should be a feather touch on the bed without triggering the probe, which would be -0.2mm on the bend.**

Instead, the probe is immediately triggered - and I have to manually adjust Z to 0.2 to untrigger it - which should be 0.0 and not 0.2.

TitusLabs, 2022-02-21:
I added some more explanation on the switch_offset and why someone would demand for a negative value: [https://github.com/protoloft/klipper_z_calibration/tree/v0.8.1#how-about-a-negative-switch-offset](https://github.com/protoloft/klipper_z_calibration/tree/v0.8.1#how-about-a-negative-switch-offset)

closed 2022-03-26 by TitusLabs