protoloft/klipper_z_calibration #29: Understandig problems aka brain goes afk

wuast94, 2021-12-16:
I have a question and hope it's not stupid, the nozzle and the probe (wich is a endstop if I'm right) are fixed, the probe needs to be lower than the nozzle then. That means that the probe is constantly scratching over the print bed and printed parts (if not using z lift)

Where is the point where my mind goes lost there?

BTW I have a ratrig v-core3 with a bltouch and from time to time my z offset gets wrong without changing anything and now I am searching for a solution to automate this process as u did. I hope someone can answere my questions :)

eduncan911, 2021-12-25:
Ah, the brain is unaware of the Klicky Probe options that Vorons and Annex Engineering uses.  Annex calls theirs the "quick draw" endstop.

It's a magnetic probe that gets "picked up" during homing, it touches the bed as it's lower than the nozzle (hence probe offset in the renderings).  And the probe gets dropped off when done.

Quick video showing it.  It's hanging down off the gantry to the left in this video: https://youtube.com/watch?v=FXPBqCQUCng

Ok the README, you'll see the nozzle endstop, and then the large rectangular object - that's the Klicky Probe with a tiny microswitch on it.

TitusLabs, 2022-02-19:
@eduncan911 thanks for your nice explanations :+1: 

This approach eliminates nearly everything which could change in a printer and would also lead to a changing Z offset. The good thing of the simple dump switch is, that it does not suffer from any temperature drifts like metal rods or electronics do. If you print in a closed printer with a bed temperature of 100C and a chamber temperature of up to 60C, this does have a big impact.

closed 2022-02-19 by TitusLabs