protoloft/klipper_z_calibration #30: For Load Cell/Nozzle probes, How to adjust for the probe's Z offset?

eduncan911, 2021-12-26:
I am attempting to use this plugin with a machine that has a load-cell on the nozzle, using the nozzle itself as the probe for perfect 0.0 offsets.  As well as a dedicated Optical Z endstop.  This is similar, but different than #28.

The problem I'm having is that the load cell has a flex that equates to 0.2mm at the nozzle.  This is a measurement that cannot be "measured" automatically via this script as this script attempts to use the klicky probe body to find the offset from the switch to the switches body.

Unfortunately, with a load cell there is no "switch body" to measure again - the nozzle itself is 0.0mm.  However, with the load cell's flex, the nozzle is -0.2mm from the trigger.

I previously had `probe -> z_offset -> -0.2` and this worked fine before this script was added.  

Now, I am unable to set any firm offset what-so-ever, in `switch_offset` nor `z_offset`.  Also, `SET_GCODE_OFFSET Z_ADJUST=x` doesn't seem to be working even setting it after a `CALIBRATE_Z`.

---

Ps, any way to remove the Nozzle and Switch probing?  Those are always directly equal to my Z endstop, because there is nothing to probe on a load_cell setup: just touch the nozzle, and that's 0.0 immediately.

eduncan911, 2021-12-27:
I ended up using `SET_GCODE_OFFSET Z_ADJUST=-0.2` in my `CALIBRATE_Z` macro override, at the end.  Turns out I wasn't verifying with `GET_POSITION` for the `gcode_base` variable, which did show `SET_GCODE_OFFSET Z_ADJUST` working.

As for skipping the switch probe, I had to edit the `z_calibrate.py` to:

    # switch_zero = self._probe_on_z_endstop(self.helper.probe_switch_site)
    switch_zero = nozzle_zero

I wonder if this could submit this as a Pull Request?  The possibility of skipping the Switch Probing step by having it set to the same as the Z endstop?

TitusLabs, 2022-02-19:
I'm not sure why or how you use this extension. If you have a load cell at your nozzle, then this extension would be pointless since you already have a direct nozzle-probe :thinking: 
So, this flex means that you need to over travel by 0.2 until the sensor reacts? Is this a fixed or varying value?

eduncan911, 2022-02-19:
so basically, i'm cheating...  i;m using this plugin to take two measurements, and then the negative -0.02 to account for the load cell.

I got around all of that, but commenting out the line above.  And I suggested making a PR to add that as a new variable in the ocnfig, that can be flipped on/off (defaulting to False, so it's backwards compatible).

TitusLabs, 2022-02-20:
This is a nice creative solution!
But, since you misuse this plugin, I will not introduce an option to disable the probing on the switch. This step is essential for the intended purpose.

The core function is just the probing on the endstop. Maybe Klipper could be more flexible on this point. Then a little macro would be enough to achieve this  :thinking:

eduncan911, 2022-02-23:
Aiight, if no option to set or override `switch_zero`, then I'll close the issue.  Thanks again for looking!

closed 2022-02-23 by eduncan911