protoloft/klipper_z_calibration #34: Why is probing surface necessary for z calibration

pfn, 2022-02-06:
> Determine the height of the print surface by probing one point with the mag-probe.

Reading this part is not clear as it appears entirely superfluous. Z endstop probe followed by clicky body probe gives offset already. An additional surface probe doesn't add anything, unless it is intended to calibrate the offset of the Z endstop, which shouldn't be used for homing anyway...

zellneralex, 2022-02-12:
Many of use have more than one flexplate aka print surface, furthermore there is thermal explanation of your bed. To get these 2 factors in the measurement you need to probe the surface.

pfn, 2022-02-12:
Clicky offset is the only probe that matters. Any homing probe should be done with clicky, IMO, not the Z endstop. Deriving a Z offset for the Z endstop is nonsense as it is a superfluous item.

zech, 2022-02-17:
I think @pfn has a point here.

But that would require to define `[probe]` as a virtual Z endstop.
That way the endstop would only be used to find the offset between nozzle and switch but still including the tunable switch offset to define the squish you want.

The different build plates would be handled by probing the surface directly to find our 0 because it's defined as a virtual Z endstop.

Unless there's something he's missing (or I am) I'd think that's correct and it would eliminate one step and maybe some variance as well.

And we only need to find this offset once per hotend/nozzle change and maybe temperature change if we'd consider thermal expansion of the nozzle (that maybe a reason for the current procedure alone).

It's also possible that defining the probe as a virtual Z endstop could disable the physical Z endstop. I guess that would require some more "modding" in Klipper to get both the virtual Z endstop and the physical one as a measuring device.

pfn, 2022-02-18:
I mostly want to understand why the third probe is necessary, but difficulty with klipper would explain the reasoning. I have my own implementation of this that doesn't require the last bed probe, but it's not on klipper.

I wanted to make sure I wasn't missing anything with regard to what opportunities I would miss without the last bed check. It seems like there isn't anything for my configuration.

pfn, 2022-02-19:
> What is your implementation? Is it public?

@TitusLabs my implementation is https://github.com/pfn/voron2-rrf-config/blob/master/macros/Calibration/Calibrate%20Z%20Offset -- plain and straightforward.

TitusLabs, 2022-02-21:
Oh nice, an other implementation for RRF :ok_hand: It seems to be so much easier ... but it's so unreadable :exploding_head:

pfn, 2022-02-21:
> Oh nice, an other implementation for RRF 👌 It seems to be so much easier ... but it's so unreadable 🤯

I get what you're saying. I read gcode fluently as a result of using rrf. Sucks for anyone that doesn't though.

TitusLabs, 2022-03-26:
I think, I need to play with RRF again on my next printer. RRF2 and DuetWifi2 times are long ago..
I will close this now

closed 2022-03-26 by TitusLabs