protoloft/klipper_z_calibration #36: Problems to install klipper_z_calibration on newest version klicky probe

Sebbeben, 2022-02-17:
hi i have been trying to install the klipper_z_calibration for days, read a lot on the repo and elsewhere, watched videos and talking to people about this. 

but i cant get it to work. it seems like klicky probe has changed their files so when i install this (klipper_z_calibration i have new version files and old version files and cant use the auto z calibration? something is not connected it seems like.

is there some way i can read about how to fix this? or will there be an update?

thanks for your good work :) this seems like a must have plugin :)

Sebbeben, 2022-02-19:
> Many thanks!
> 
> This Plugin/Extension is not related to the Klicky Probe. Just use the Klicky with it's Macros. Then you install this extension and add the correct Macro names to the `start_gcode` or `before_switch_gcode` and `end_gcode`. That's it!

Thanks for your reply :) I will try this

Sebbeben, 2022-03-04:
PROBLEM SOLVED:

i got help from chungr

this was my solution:
here are the steps:
a) rename klicky-z-calibration.cfg to old-z-calibration.cfg - you can do this by right clicking and you should see a RENAME option in Mainsail
b) use create a file (the icon that has a + sign on a file or 2nd icon from the left at the top) and call it klicky-z-calibration.cfg in Mainsail
c) go to Klicky github, in Klicky macros, click on klicky-z-calibration.cfg
d) copy the content and paste them into the klicky-z-calibration.cfg in Mainsail
e) enter the values that you have for your printer in the z_calibration section (the values are now in old-z-calibration.cfg)
f) save and restart

Now you have the correct klicky-z-calibration.cfg.

TitusLabs, 2022-03-26:
Great!
Meanwhile, I removed by example scripts and updated the configuration section to use the Klicky configs.
Then, I will close this now..

closed 2022-03-26 by TitusLabs