protoloft/klipper_z_calibration #38: Returning probe nozzle crash in the bed

acosano, 2022-02-20:
Hi. I'm using Euclid with the plugin. If  I do manual attach and detach of the probe it works fine. If  I do a QGL it works fine but as soon I do  Z calibration it will pickup the probe do the z offset and when it put it back it will be always hit the bed with the nozzle or the probe will not return properly into the dock. the question is why? if it can't pick it up and using the same coordinate in reverse mode it should bring it back as the dock location or position has not change. Any idea what I'm doing wrong?

TitusLabs, 2022-02-20:
First of all, with a magnetic probe you need to set a higher clearance (z_offset, horizontal_move_z) in the probe, quad_gantry_level and mesh sections. The z_calibration section has a clearance property too.

Is your Z=0 after first homing nearly the same to the bed surface? What is the calculated offset after calibrating Z?
And how is it able to hit the bed with the nozzle when the probe is still attached? What does this macro? Be careful with the save and restore gcode state statements if you really want so change this state. Is your dock mounted to the bed?

acosano, 2022-02-20:
I was able to sorted it out  but in a unconditional way. I created a additional detach macro where I change the Z dock location to a higher position. that fix the problem but what  I don't understand is the dock is mount to the bed, the bed is fix; so the dock position and coordinates never change. So why it can pick it up but not return to the same place after doing the Z calibration. The Z pickup is -0.68  but after calibration I need to use 3.68 to be able to dock it back.

Is there away to tell to not apply changes to the z location of the dock?

TitusLabs, 2022-02-21:
You found the only disadvantage of attaching the dock to the bed!

You can do some things to prevent this:
- Configure your endstop position to keep the measured z offset small
- Do a G28 Z after a QGL/Z-Tilt to keep the measured z offset small
- To not drive into the bed, you can use a higher clearance value
- Chose good values for the attach/detach process - it can handle some variations
- Keep in mind what SAVE/RESTORE_GCODE_STATE in macros does

And sometimes it cannot be prevented at all... e.g. due to plastic at the nozzle and first homing in cold state - then you are too high to fetch the probe.

TitusLabs, 2022-03-26:
I will close this now!

closed 2022-03-26 by TitusLabs