protoloft/klipper_z_calibration #39: Configuration is too complex

rlaferla, 2022-02-20:
I was hoping this plugin would solve Z offset once and for all for my Voron Trident.  I installed it but then I started to configure it and there a tons of parameters that need to be set.  I think this is a good project but it needs more work.  It could extract the z_safe_home values for the position of the nozzle.  It could use the Afterburner or Stealthburner distance between nozzle and probe as a default or as  comments in the cfg.  There must be ways of cutting down the sheer number of these parameters.

TitusLabs, 2022-02-20:
Most of the properties are predefined with the values taken from all the other objects in Klipper. Someone could argue, that this is not a best practice too.
All defaults are documented. Even the bed site defaults to the relative reference index of the bed mesh - if configured.
And generally, this is a Klipper extension - not a Voron extension.

Do you have any concrete improvements?

rlaferla, 2022-02-20:
I am providing constructive criticism to improve the plugin.  That is why I wrote "I think this is a good project".  If it is a general Klipper plugin, that's great but somehow I misinterpreted the description.

[image]

rlaferla, 2022-02-20:
As for more concrete improvements, perhaps printer specific .cfg templates?  Perhaps based on popular sizes with values for different sizes in comments?  e.g.  # 350mm use x, 250mm use ___, 300mm use ___

TitusLabs, 2022-02-20:
Thanks and don't get me wrong - I know that the documentation is complex :see_no_evil: and every improvement is welcome!

Oh, but this is a good point. I will change the description..

I don't think templates are necessary. Where possible, default values are provided and I cannot provide any predefined positions in the configuration. But maybe I can add a minimal configuration section with all mandatory properties :thinking: 

I am not sure about the example configurations though. These are only examples and maybe they make it worse. They are all about configuring a Voron using a mag-probe. For the z calibration, there is only a configuration section and a macro... By deleting them all, I would refer to the Klicky configuration/macros (as I already did in the documentation) and the all in one config from zellneralex...

TitusLabs, 2022-03-26:
I update the configuration section and added a minimal configuration example. The example scripts are removed completely without replacement. There is also a PR pending which reduces the x and y position properties to one x,y coordinate property.

If there are no more suggestions, I will close this ticket...