protoloft/klipper_z_calibration #40: Since update not working anymore

mochoandre, 2022-02-21:
hello, I have updated z_calibration trough moonraker 4 or 5 days ago I it will not work anymore.
First without changing old configs, when printer was doing some calibration routines at start of a printer it would do  like home but after when starting z_calbration looks like the two AB motors are working against each other with a rattling sound.

Try to use your new macros and configs but first it was heating my nozzle to 220 and bed to 100 without my gcode file need or want it. And also had crash issues the probe with z_endstop pin and had to increase clearance by a lot. but still not working.

When it starts to print imidiatly starts to loose steps. I gess it's it's because how your macros setc current to the motors.

Tryed to use Kliky examples and also could not make it work.

Now I have a very big paperweight.

regards

mochoandre, 2022-02-21:
Also, When I first installed documentation between Klikky and Z_calibration was somewaht "heavy" but consistent!! Now with this updates it's a very big mess

TitusLabs, 2022-02-22:
I'm sorry to hear this! But I cannot believe that the z_calibration update is the cause of this...

Maybe you also updated Klipper? Did you forgot to update the firmware on the MCU(s)? This is very important!

And don't just copy any macros if you don't understand them. You even don't need my macros - they are just examples. If you have a working setup, there is no need to update any of your macros. This is all documented!

What consistency do you mean? I didn't change much. It's all the same as before. And be more precise about your "big mess", if you want some help or use the community at Discord.

This is all done in my spare time and you don't believe how much time this is!

TitusLabs, 2022-03-26:
Then, I will close this now..

closed 2022-03-26 by TitusLabs