protoloft/klipper_z_calibration #42: Not an issue, but an idea about autocalibrating switch_offset value

falcon14141, 2022-03-07:
Hi,

I successfully use your code on my Voron2 equipped with a klicky probe.
Manual calibration of the switch_offset works fine, but is still a hassle.

I recently had to change my klicky probe to a spare one, and unfortunately, the switch_offset wasn't  the same on the 2nd microswitch.

This gave me an idea on a possible way to try, and maybe achieve autocalibration of the swith offset:

On a voron we have to "endstops" being the z_endstop end the probe.
If we probe the magprobe two times, we could maybe get the relative distance travelled by the microswitch.

I'll try to explain my idea further with the description of a calibration sequence:

- probe nozzle on Z-endstop
- attach magprobe
- probe magprobe main microswitch body on Z-endstop
- probe magprobe microswitch actuator on Z-endstop and wait for habing both Z_endstop and probe endstop triggered

We should now have the meens to compute both relative distances between all required values.

Unofrtunately, I lack the base knowledge on klipper to implement this idea, and would appreciate your insights

gongloo, 2022-03-10:
I was just thinking the same thing, but a step further: if we probe the two end stops right on top of each other as you describe, then we don't need to probe the body of the mag probe at all. Thoughts?

TitusLabs, 2022-03-25:
I'm sorry, this will not work! Please read the switch section. I had this discussion already many times!
How would this solve your intention to measure or eliminate the switch offset? Can you tell what exactly would happen if you do so? I cannot.. it depends on so many things and even if it does what you think it can be very fragile. It would just work out in a perfect world where both switches always trigger at the exact same time..

jangrewe, 2022-06-22:
I had the same idea, but just because the Z Endstop's switch triggers, doesn't mean that it doesn't travel further down after being triggered - potentially falsifying the actual travel distance. Same for the magprobe. Depends on who triggers easier.

This would (maybe) work if both switch trigger and have 0 travel of the switch actuator afterwards... or not, i need to read up on that again. ;-)

closed 2022-07-07 by TitusLabs