protoloft/klipper_z_calibration #53: Install script not working

Lukas-Sturm, 2022-04-10:
Hi I am not a linux pro.

But after running the new installer (updated to 0.9) the created link in `klippy/extras` points to `//z_calibration.py`. And Klipper fails when restarting the firmware.

I changed it manually to point to the right file, now it works.

The produced output of the `install.sh` (I have no idea on how to interpret this :) )
```
install.sh: 63: install.sh: Bad substitution
install.sh: 53: [: Illegal number:
Linking extension to Klipper...
Restarting Klipper...
```

_Also wondering if there is some kind of Discord or some other way of discussing ( have some questions about accuracy)_

TitusLabs, 2022-04-14:
This is not an issue with the recent update... but this is strange, in what environment did you execute it?

Lukas-Sturm, 2022-04-15:
`ssh`ed to the PI running mainsailOS (roughly 4-5 months old) and executed it from the home directory /home/pi. 

The install script uses two variables, so I tested the creation in the shell by just pasting the definitions and echoing them, they turned out to have the right values.

TitusLabs, 2022-04-17:
You are the first with this problem. And I have no idea at the moment.. but, you can find me on the Voron Discord if you want.

Lukas-Sturm, 2022-04-19:
I will close this then, I fixed it on my machine by manually linking.
Probably just a combination of miss configurations on my side :)

closed 2022-04-19 by Lukas-Sturm