protoloft/klipper_z_calibration #56: Slow down the last probe action during z calibration

Nakazen, 2022-04-17:
Hi,

I am currently successfully using your awesome plugin in and I am absolutely loving the function.
I have just one issue with the plugin and I can't seem to fix the issue myself.
The plugin has been setup in such a way that it probes the nozzle very slowly on the z switch 10x, the attached dock on the z switch 10x and then it moves to the middle and it probes 2x times at a much higher speed. (It looks like the same speed I use for the bed calibration)
I would like it to slow down and use the settings that are used for the other probes.

Is that something the plugin can do and I have incorrectly configured it or is that not an option?
I have attached my config

[z_calibration(6).txt](https://github.com/protoloft/klipper_z_calibration/files/8501579/z_calibration.6.txt)
Please let me know.

TitusLabs, 2022-04-17:
Hi, nice to hear that you like it! But I'm sorry, this is not possible. Please read the Info at the end of the configuration section. It just triggers the normal probing routine there. So yes, these are the speed of your probe section. Different speeds are not possible there. And I'm not sure if it's so easy to change this because the probe is completely different to the z endstop :thinking:

TitusLabs, 2022-07-08:
Sorry, I was wrong! I have implemented a complete probing routine for the Z endstop. So, I can adapt this using the probe too. Thanks for the hint in the right direction :exploding_head:
A new release is on it's way...

TitusLabs, 2022-07-11:
This should be solved now with the release 0.9.2.

closed 2022-07-11 by TitusLabs