protoloft/klipper_z_calibration #57: Offset measurement goes bad every few prints

Silikowski, 2022-04-17:
Hi - it might be more of a Userquestion than an issue - i use the plugin and it mostly works fine. But somehow (my pei sheet tells a few stories already) the offset changes so much, that it scars the sheet.

I do very simple startscripts:

`[gcode_macro G32]
gcode:
    G28
    G0 Y330 F3000
    CLEAN_NOZZLE
    G28
    QUAD_GANTRY_LEVEL
    CLEAN_NOZZLE
    CALIBRATE_Z
    ##  Uncomment for for your size printer:
    #--------------------------------------------------------------------
    ##  Uncomment for 250mm build
    #G0 X125 Y125 Z30 F3600
    
    ##  Uncomment for 300 build
    #G0 X150 Y150 Z30 F3600
    
    ##  Uncomment for 350mm build
    G0 X175 Y175 Z30 F3600
    #--------------------------------------------------------------------
   
[gcode_macro PRINT_START]
#   Use PRINT_START for the slicer starting script - please customise for your slicer of choice
gcode:
    G32                            ; home all axes
    BED_MESH_CALIBRATE
    G1 Z20 F3000                   ; move nozzle away from bed`

Home, clean nozzle, home again, qgl (its a voron 2.4), clean nozzle again, calibrate Z.

Its fine - my values are consistent, no big spreads (single jumps but median is fine)

`probe at 175.000,155.250 is z=7.608202
probe at 175.000,155.250 is z=7.610702
probe at 175.000,155.250 is z=7.610702
Switch body >>
probe at 228.000,333.000 is z=8.104452
probe at 228.000,333.000 is z=8.104452
probe at 228.000,333.000 is z=8.104452
probe at 228.000,333.000 is z=8.104452
probe at 228.000,333.000 is z=8.105702
probe at 228.000,333.000 is z=8.105702
Nozzle >>
probe at 231.000,350.000 is z=0.003202
probe at 231.000,350.000 is z=0.001952
probe at 231.000,350.000 is z=0.000702
probe at 231.000,350.000 is z=0.000702
probe at 231.000,350.000 is z=-0.000548
probe at 231.000,350.000 is z=-0.000548
`

But my overall offset changes from .55 to .8 from print to print, beeing to close from time to time.
Now i changed the nozzle and the measurement (i posted above) gave me an offset like OFFSET=-0.823048 what was .35 to big. i needed to adjust it to around .5 to start printing and stopping killing my pei sheet.

the plugin should calculate the offset even after changes of builplate or nozzle right? is there anything i can do to get consistent offsets? I adjusted my switch offset so it did a good job on the first sanity test and then said "fine, thats it" and hoped from now on it should do the trick. my sheet tells another story.
magnets are glued into place, no movement here so the klicky should be fine. should i do more probes?

thx!

TitusLabs, 2022-04-17:
Yes, it should be "fine, thats it"...

First, I would change the order to: QGL, G28, clean nozzle, calibrate_z

But, it sounds that there is something wrong in your system. You need to check everything! Is there a loose pulley on the z drives? Is your probe ok? Do you use an original Omron switch? Are the probing positions correct - especially on the switch? Do you probe slowly multiple time and use median as samples result? You can also use PROBE_ACCURACY and PROBE_Z_ACCURACY...

Silikowski, 2022-04-18:
my z idlers are fine as far as i can see.
probe accuracy tests seems ok too:

probe accuracy results: maximum 6.988750, minimum 6.972500, range 0.016250, average 6.978750, median 6.977500, standard deviation 0.006175

i dont have an original omron switch cause they were not in stock somewhere.

probing positions are fine. i tuned them by hand and checked the probing Multiple times.

im gonna change my starr order as proposed and get back to you

Silikowski, 2022-04-19:
Well i replaced my start_print macro with the new order and followed a few hints from the voron discord. it looks better now (so far) - im gonna let it run a while and see if stays fine consistently

closed 2022-04-19 by Silikowski