protoloft/klipper_z_calibration #59: Reorder probe sequence to reduce ooze

poernahi, 2022-04-30:
Heating the nozzle at print temperature for extended period of time leads to unpredictable amount of ooze. The ooze can sometimes interfere with various probing operations. One way to reliably reduce ooze is to only heat the nozzle to slightly below the ooze temperature (let's call it warm nozzle) during most probing operations and do a final z-endstop touch at printing temperature.

With the current sequence of probing nozzle -> switch body -> bed, the nozzle has to be at printing temperature throughout the entire calibration process, and may ooze during bed calibration. 

How about reordering the calibration:
- Initial homing, QGL / tilt adjust, re-home z (warm nozzle)
- Probe switch body (warm nozzle)
- Probe bed (warm nozzle)
- Dock probe
- Heat up nozzle
- Nozzle cleaning can be done here
- Probe nozzle
- Print

It may be necessary to do a final Z homing with the hot nozzle right before printing.

TitusLabs, 2022-07-07:
Yes, that could be a good sequence. I mean, I don't have a problem with oozing. I do clean the nozzle and the next step (and first step of the calibration) is probing the nozzle. So, oozing afterwards is not an issue anymore. This would be cleaned with a intro line (or brim) before starting the print. But, there maybe different situations out there...
The problem is, you split the calibration process into two parts. And I don't want to do that. That would be too complicated and error prone. The alternative would be to integrate all the steps between as custom gcode... and I don't like this as well :thinking: 
How is it now? Is oozing still a problem for you?

poernahi, 2022-07-23:
After running further experiments with various ooze mitigation strategies, I found that the most repeatable process is the following:
- At print end or print cancel, retract enough filament to clear the nozzle. For my setup this is approximately 10mm.
- When preheating before print, set hotend to a temperature where filament ooze easily. We want as low viscosity as possible but not too hot that it burns the filament.
- Wait until ooze stops. The nozzle should be empty now.
- Clean up the nozzle
- Probe and calibrate. There should be no more ooze at this time. Can even lower the hotend temperature a bit here.

Without clearing the nozzle, I found that some of the more viscous material like PETG creates a very noisy nozzle height reading. Modifying the z-endstop to require higher actuation force helps somewhat but still unreliable as some plastic can still get between the nozzle and endstop pin. There is a limit on increasing the actuation force as well as too much will cause deflection of the whole toolhead depending on the stiffness (quality) of the X linear rail and the toolhead assembly. 

I will close this for now as my idea above is not good enough to solve the oozing problem. Thanks for the plugin!

closed 2022-07-23 by poernahi

TitusLabs, 2022-07-23:
Great! I heard about ooze problems especially with the Revo hotend. Waiting some time helps :+1: