protoloft/klipper_z_calibration #62: Attach & Detach Probe Documentation Macro has simple spelling mistake

evader, 2022-05-10:
Love your solution. Simple mistake under documentation for Attach & Detach Probe Macro on your main page.
You have CG28 instead of G28. Hope this helps and many many thanks for your tremendous work.

[image]

Hoping the macro should be like the following.

[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    {% set bed_position = params.BED_POSITION|default('None') %}
    G28 
    M117 Z-Calibration..
    ATTACH_PROBE                # a macro for fetching the probe first
    {% if bed_position != 'None' %}
      BASE_CALIBRATE_Z BED_POSITION={bed_position}
    {% else %}
      BASE_CALIBRATE_Z
    {% endif %}
    DETACH_PROBE                # and parking it afterwards (or DOCK_PROBE in klicky macros)
    M117

jangrewe, 2022-06-22:
It is: https://docs.vorondesign.com/community/macros/macros/120decibell.html

TitusLabs, 2022-07-07:
Yes, that is the conditional homing macro. But sorry, that is not clear in this context... I will change that :+1:

closed 2022-07-07 by TitusLabs