protoloft/klipper_z_calibration #71: Z_OFFSET_APPLY_CALIBRATOR

charlespick, 2022-09-24:
Idea for a gcode that works like Z_OFFSET_APPLY_PROBE/ENDSTOP that would apply the current offset to the switch offset for fast setup.

TitusLabs, 2022-11-02:
So, you mean to save the determined nozzle-switch-offset permanently? Currently, the problem is that the script is not able to handle the probe used as a virtual endstop..

charlespick, 2022-11-07:
@TitusLabs you wouldn't really use it to save the calibrated offset. You could use this by restarting the printer to clear all offsets, home, calibrate z, save that to the endstop, then restart, home, and use the manual controls to bring the nozzle to the bed at z=0, then apply the remaining difference to the switch-offset. This way you can get your switch offset close without doing a ton of math.

TitusLabs, 2023-03-12:
I will add a hint for a new calculated z endstop_position which can be used as a reference for adjusting the configuration. Is this what you mean?

charlespick, 2023-03-15:
it should be a new calculated switch_offset value, nothing to do with the endstop.

TitusLabs, 2023-03-22:
Ok, I think I got it. I'm already testing it... :crossed_fingers:

TitusLabs, 2023-04-25:
I have added a new CALCULATE_SWITCH_OFFSET which tries to determine a reasonable switch offset. Is that how you imagined it?

closed 2023-05-25 by TitusLabs

charlespick, 2023-05-30:
> I have added a new CALCULATE_SWITCH_OFFSET which tries to determine a reasonable switch offset. Is that how you imagined it?

probably. my voron is broken rn so I can't test.