protoloft/klipper_z_calibration #73: CALIBRATE_Z decreases z incrementally

tloesch, 2022-10-12:
When using CALIBRATE_Z at the end of my PRINT_START macro the nozzle will move closer to the bed incrementally.
Observable by a decreasing line height and an increasing line width.
When i remove CALIBRATE_Z from my PRINT_START macro the issue is gone.
Even when i set the same z-offset (previously calculated by CALIBRATE_Z) the issue is gone.

Am i doing something wrong or is it a bug within CALIBRATE_Z?

Iam at version v0.9.2-0-gf212b40.

garethky, 2022-10-23:
Do you mean that running multiple prints back-to-back results in a lower nozzle each time?

This plugin resets the absolute offset to 0 and then uses the `Z_ADJUST` parameter to apply the measured offset. See: https://github.com/protoloft/klipper_z_calibration/blob/f212b409cd1728b8edaabc7ec0f6144542dd5356/z_calibration.py#L313-L323

The plugin shouldn't be the cause of the behavior you're describing.

* Are you just seeing variance print-to-print? Does the nozzle eventually hit the build plate or does it stabilize?
* Is there anything else in your config that calls or overrides `SET_GCODE_OFFSET`?

tloesch, 2022-10-30:
i also looked into the plugin code and was also wondering what the issue could be.
i checked my complete printer the last two weeks and found an issue.
fixing this issue fixed also the problem. so its not the fault of this plugin. which should be obv when looking into the source.

the issue was that the main wire on my toolhead board (provided by ldo) was slightly loose.
i put it in correctly and the issue was gone.
i have klicky running on my toolhead.
maybe the probe couldnt return correct values because the wire was slightly loose.

closed 2022-10-30 by tloesch