protoloft/klipper_z_calibration #74: Klipper crash immediately after stowing euclid probe - Doesn't happen with just calling M402

JohnGalt1717, 2022-10-21:
When I call CALIBRATE_Z it does it's thing, gets the probe, does the calibration, returns the results, then does the m402 and returns the probe.

As soon as it's done, klipper crashes.

Not sure if this is an auto_z issue or a klipper issue. I didn't see anything other than my canbus toolhead became not-resonsive after the calibrate_z completed?

I'm using an Octopus 1.1 with EBB-36 with canbus passthrough for the Ocotpus (i.e. the pi is connected to the octopus with usb c and the octopus is connected to the ebb-36 over the canbus RJ-12 connector)

Here's the logs:
[klippy (2).log](https://github.com/protoloft/klipper_z_calibration/files/9842572/klippy.2.log)

TitusLabs, 2022-11-02:
Hi, there is no error concerning the calibration script. But, in line 80751 it says: Timeout with MCU 'EBBCan'
I have never used CAN Bus passthrough or even the EBB-36. There can be many reasons for this timeout. But, I have observed, that the probing is time critical and can consume more resources (or something else). I recommend a Pi4 with CAN Bus. If you use KlipperScreen, stop it and try it again...

TitusLabs, 2023-03-12:
Any news on this? Otherwise I will close this ticket...

closed 2023-03-20 by TitusLabs

JohnGalt1717, 2023-03-21:
Sorry, I switched to Voron TAP so I didn't need this and for some reason this notification got shunted to spam. Good to be closed.