protoloft/klipper_z_calibration #75: Clearing up some confusion

AndrewEllis93, 2022-11-08:
We frequently have users in the Voron Discord who are very confused by how this Z calibration works. 

Particularly users who have no idea that they need a good base Z calibration first, and will set the `max_deviation` very high instead. Then, if calibration ever fails, they will dig a trench through their PEI.

I think it would clear up a lot of confusion if it were made clear in the documentation/naming that it's auto *baby stepping* rather than auto *calibration*.  That might indicate to more people that they need to still use Z_ENDSTOP_CALIBRATE.

TitusLabs, 2023-03-12:
It's finally on it's way... :+1:

TitusLabs, 2023-04-25:
Finally.. done!

closed 2023-04-25 by TitusLabs