protoloft/klipper_z_calibration #77: Section 'z_calibration' is not a valid config section

76900289, 2022-12-25:
klipper      v0.11.0-41-g9b60daf6
Section 'z_calibration' is not a valid config section?

gdachs, 2022-12-25:
That is not an issue. Your configuration is wrong. Most likely you didn't add `[include z_calibration.cfg]` to your printer.cfg.
Before you open an issue on github, always triple check that the problem is not on your side.

76900289, 2022-12-25:
Added. It is a problem after updating KLIPPER

76900289, 2022-12-25:
It is normal before updating

gdachs, 2022-12-25:
I use the same Klipper version with z_calibration, no problems. Upload your klippy.log.

76900289, 2022-12-25:
[klippy.txt](https://github.com/protoloft/klipper_z_calibration/files/10299742/klippy.txt)

gdachs, 2022-12-25:
what gives `ls -l ~/klipper/klippy/extras/z_calibration.py`

76900289, 2022-12-25:
I see. The update uninstalled this file. Thank you very much

closed 2022-12-25 by 76900289

gdachs, 2022-12-25:
How did you update? Anyway, so you can close this issue.

76900289, 2022-12-25:
[image]
here

aCtionBR, 2023-06-04:
Hello I have just executed an update all in my configuration manager. I had my z_calibration working perfect before and now the system is presenting this same error after this update. I am adding a print of the error and a copy of klippy.log. 
Can you please help me to identify what is wrong?

[image]

[klippy.log](https://github.com/protoloft/klipper_z_calibration/files/11646567/klippy.log)

aCtionBR, 2023-06-04:
I solved it. 
I have seen a lot of comments in other foruns about this problem, but none of them commented what solved my problem, so I am adding below what was the problem. Maybe it helps others.
There has been an update on the format accepted in z_calibration.cfg file. I have not looked into when it happend, what I learned is that the old configuration directives (like probe_nozzle_x: xxx) is no longer accepted. You have to adapt your z_calibration.cfg file to the new standard that uses nozzle_xy_position insted of  probe_nozzle_x and  probe_nozzle_y.

I got it from the configuration file presented at https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Configure-It. Then I researched for some examples of z_calibration.cfg file in the internet to double check.