protoloft/klipper_z_calibration #78: offset calculate

Reneeetje, 2023-01-01:
I have a higher endstop, like 8 mm. from the bed.
I cannot get auto z working like normal it give a large offset, so i  can not usethis with --> max_deviation:.
now i have to do :
[stepper_z]
position_endstop: 0
the z offset is OFFSET=-7.114625, but i cant use max deviation, the deviation is larger dan 7 !

Is there a way to to compensate the endstop height.? I cannot, move the enstop because i have a ender 5 plus, guidebars are in the way !

Nelus82, 2023-01-26:
I'm having the same problem. My endstop is the same height as my bed (position_endstop = 0).
Everytime when I do a Z calibration it gives me a Z-Offset: 1.2

The probe z_offset does nothing.

RcTomcat1, 2023-01-31:
I am experiencing the same issues after updating klipper and moonraker to the latest versions...Yesterday my printer was doing fine, today it crashes the nozzle

LaRenard, 2023-02-22:
I've recently added a mod to a printer and set up auto-z but it appears to be doing the same behaviour described here, my endstop is 4mm above the bed which I've set in stepper_z config
Manually doing the probe calibration I get 4.x mm which I added to my probe config but the auto-z is always touching the bed when the print starts:

`Z-CALIBRATION: ENDSTOP=4.015 NOZZLE=4.012 SWITCH=8.088 PROBE=4.663 --> OFFSET=0.137500`

TitusLabs, 2023-03-11:
Have you been able to solve this in the meantime?

I have not observed such behavior with my printers. Especially since the plugin should be very independent from the rest of Klipper at its core. And it can't be related to a Moonraker update.

A correct auto calibration does not depend on the values z endstop position or z probe offset - although it is advisable to set them reasonably accurately.

Nelus82, 2023-03-11:
Yes, the problem was the heightmap. It was 1.2 mm off. I had to create a good mesh at 0 (with probe z_offset). And then the z calibration is almost perfect. 

You can also disable auto load mesh.

TitusLabs, 2023-03-12:
Ok, great! Most problems are related to bed mesh. Then I will close this ticket..

closed 2023-03-12 by TitusLabs