protoloft/klipper_z_calibration #81: Bed_mesh_calibrate immediately before or after calibrate_z

stahlfabrik, 2023-03-09:
Hi!

thanks for your awesome Klipper „plug-in“!

After reading the readme I am still unclear if I run the bed_mesh_calibrate after the calibrate_z or just before calibrate_z?

could you please help me figure this out?

also, this might be useful info in the readme 😉 for others to find. 

Best regards

stahlfabrik, 2023-03-09:
One further comment: as I understand your calibrate_z Macro does work with adaptive bed mesh calibrate where a smaller region on the bed is only measured. 
thus I would conclude that calibrate z needs to be performed AFTER bed mesh calibrate - otherwise you would not know the reference point?

But I have seen print_start macros where the bed mesh is done after calibrate z. So I am lost and confused 😀

please help! 🛟

stahlfabrik, 2023-03-09:
Okay I played some more with it and did some more reading. As the plug-in „just“ sets the gcode z offset it should be fine to run it last - after bed mesh calibrate. Do you agree with my conclusion?

TitusLabs, 2023-03-09:
Hi, it should make no difference if you use the relative reference point of the mesh. At that point, the mesh will always be 0 in z because that is the reference. And that will be calibrated before or after the mesh. But as you mentioned, if you are using an adaptive mesh macro, you should calibrate after the mesh.
Thanks for the hint. I will add this to the start sequence...

dmnkhhn, 2023-03-20:
So in this case _bed_mesh_calibrate_ needs to be run before _calibrate_z_?

TitusLabs, 2023-03-20:
If you change the relative reference point of your mesh, then yes. You should calibrate always at that point. If you do not configure a probing point on the bed, it will read the current relative reference point before probing at it.

closed 2023-04-25 by TitusLabs