protoloft/klipper_z_calibration #82: Offset to low

MrHardstyle, 2023-03-10:
I have one Pi and two Ender 3 Pros and in both is Klipper. They are working in two instances. 
I have also a klicky probe on both devices and the tolarances are up to 0.05. Both printers are the same Setup and both are in Same config. One Ender ist working Like a Charm with the Auto z but the second one is getting whole time to Low.
When i Change in the config the Offset of the Switch nothing Happens. 
My start Code is G28,G29 for BedMesh, Calibrate Z
I have No BedMesh saved and No Z Offset in the probe.
AS i told config issue is Not present because both are in the exact Same config. Only E Steps of Extruder are a bit different.
Could it be that the Plugin is Not for more than one instance of Klipper?

Klipper newest Version
Interface Fluidd
Plugin newest Version.

TitusLabs, 2023-03-11:
There is no reason why it should not work with two instances on one Pi. Even if you think everything is the same, they are still two physically different machines. If you really haven't overlooked anything in the configuration, it will probably be in the mechanics...

TitusLabs, 2023-04-25:
Any news on this? Otherwise, I will close this ticket..

closed 2023-05-08 by TitusLabs