protoloft/klipper_z_calibration #84: Auto Z Calibrate with Beacon Probe

esc247, 2023-03-22:
Hello,

I have been trying to set the Auto Z Calibrate with the Beacon Probe. I have a dual z bed assembly, and some help would be great!

I've installed the dependencies, but for some reason when I try to restart after saving the config I get the following error:

`Internal error during connect: 'BeaconProbeWrapper' object has no attribute 'sample_count'`

This is particularly weird as if I comment out the z_calibration section and use the beacon as a virtual endstop, it works. However, I just cannot get beacon to work with a physical z endstop so that I can set up auto z offset calibration.

TitusLabs, 2023-03-22:
Hi, thanks for the hint. I have not thought about this yet. The Z calibration tries to read the sample_count from the probe as fallback which the beacon does not need/use - this fails. But, it should work if you specify them directly in [z_calibration].

So, does it work if you specify the samples, samples_tolerance, sample_tolerance_retries, samples_result, lift_speed and clearance values in [z_calibration]?

I like the idea of getting the best of both worlds  :thinking:

esc247, 2023-04-02:
Hey, have not tried that. Will try it and let you know :)

jholguin7, 2023-04-08:
I would love to know if you get this working!

esc247, 2023-04-09:
I'm rebuilding my printer with alu parts, so will take a bit of time before I can get it up and running to test again. Will keep you posted.

jholguin7, 2023-04-09:
Thanks! I'll be following this

lucascarvalho, 2023-04-23:
I’ll be installing beacon soon, so I’m following this thread.

esc247, 2023-04-26:
> 

Hi, I am trying to set this up now.

I have added the following to my printer.cfg:
`[z_calibration]
#nozzle_xy_position:   <X,Y position for clicking the nozzle on the z endstop - not needed if [safe_z_home] is used>
switch_xy_offsets:    0,10 #<X,Y offsets from the nozzle position for clicking the probe's switch body on the z endstop>
#bed_xy_position:      <X,Y position for probing the bed, for instance the center point - not needed if mesh with relative_reference_index is used>
switch_offset:        0.5 #<offset of the switch trigger (read the Switch Offset section!)>
#start_gcode:          <macro name for attaching the probe>
#before_switch_gcode: <macro name for attaching the probe AFTER probing the nozzle>
#end_gcode:            <macro name for docking the probe>
samples: 1`

But I get this error:

[image]

TitusLabs, 2023-04-26:
Yes, as I already wrote, you need to define all the properties where the default values come from the probe, because this probe does not have these values. Please see my first answer..

esc247, 2023-04-26:
> Yes, as I already wrote, you need to define all the properties where the default values come from the probe, because this probe does not have these values. Please see my first answer..

Oh shit, sorry I pasted the wrong thing. I defined all the values, but still get that same error.

TitusLabs, 2023-04-28:
It will not try to read these values from the probe if you define them in your configuration.

Any positive test results on this so I can add a hint in the documentation?

esc247, 2023-05-04:
So, here is more info.

I configured my z calibration as follows:
`[z_calibration]
#nozzle_xy_position:   <X,Y position for clicking the nozzle on the z endstop - not needed if [safe_z_home] is used>
switch_xy_offsets:    0, 24.3 #<X,Y offsets from the nozzle position for clicking the probe's switch body on the z endstop>
bed_xy_position:      150, 150 #<X,Y position for probing the bed, for instance the center point - not needed if mesh with relative_reference_index is used>
switch_offset:        0.01 #<offset of the switch trigger (read the Switch Offset section!)>
#start_gcode:          <macro name for attaching the probe>
#before_switch_gcode: <macro name for attaching the probe AFTER probing the nozzle>
#end_gcode:            <macro name for docking the probe>
samples:              1
samples_tolerance:    0.01
samples_tolerance_retries: 5
samples_result: average
lift_speed: 5
safe_z_height: 5`

Now I can run calibrate_z. However, when I do, the moment the nozzle hits the endstop, I get the following error:
`Internal error on command:"CALIBRATE_Z"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown`

TitusLabs, 2023-05-08:
Wait a minute, what exactly are you trying? The z calibration needs a magnetic probe - there is no way around it.

I'm not sure what I was thinking when you started this ticket... may be, because I saw someone at that time with beacon and klicky mounted to the tool head. But, Klipper is not able to have two probes...

So, to get out of this. You need the beacon configured as probe and the Klicky needs to be in series with the z endstop... and the auto calibration must use the z endstop on the bed too :thinking:

charredchar, 2023-06-16:
I've been bounding the idea of using a Beacon probe with Auto-Z as well. The thing is, the Beacon is an inductive probe, it isn't all that dissimilar from the stock Omron probe when it comes to how Klipper handles it. This includes using it as a Virtual Z Endstop to probe the bed for Z0.
But I had the thought today... What if you attempted to trigger the probe off of the Z Endstop? It would probably work better with the Sexbolt Endstop mod, which I have. So if you define the default Z offset, like you do with any other probe, could you not use the probe itself to trigger from the microswitch endstop like you would with a physical probe? This would also work with the Omron probe for anyone that doesn't use Klicky or Euclid. Would that be possible?
Sadly I do not have the Beacon to test with.

TitusLabs, 2023-06-27:
I personally don't like the Beacon. This thing is super fast. And in the end, it's useless because it can only detect the flex plate surface and not your PEI surface. If your PEI is bad and somehow has air bubbles etc. then you've lost with it. In that respect, I'm still a fan of Klicky and very good hardware that has been conscientiously assembled. Then there is no need for mesh, no need for beacon.... but also no need for fancy macros and fancy graphics :stuck_out_tongue_closed_eyes: 

It would be possible to use both as I commented before. It just needs one line of code to be changed. But, I would not do this.

It may be possible to use this plugin with an inductive probe too. But, the problem with these probes (and also with the Beacon) is that they are really bad in measuring absolute distances because the measurement is influenced by many environmental parameters. So, the switch_offset would not be fix. And that's why this plugin was born with the first mag-probe to eliminate these drifts.

closed 2023-08-21 by TitusLabs