protoloft/klipper_z_calibration #88: Clarification on the new CALCULATE_SWITCH_OFFSET

devsfan1830, 2023-04-26:
So, noticed you just updated and added that function. Reading through the wiki here: https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calculate_switch_offset

You say to move the nozzle down until is barely touches the bed. But the switch probe hangs lower than the nozzle. Do you mean to center and lower the Z until the SWITCH barely touches? Nozzle would make sense for say, the Voron TAP, but seems like "switch" would be what you use for klicky style. Unless I am misunderstanding how this works.

devsfan1830, 2023-04-26:
Nevermind, I misunderstood initially. I get it now, manual z offset with the probe docked. The calculate command will fetch the probe and test the bed to determine the switch offset. Fine tune that value as needed.

closed 2023-04-26 by devsfan1830

TitusLabs, 2023-04-26:
Yes, you first have to run the calibration and then - without the probe - you go down with the nozzle and run the calculation. I will make this more clear in the documentation :+1: