protoloft/klipper_z_calibration #90: Message: `The resulting switch offset is negative! Either the nozzle is still too far away or something else is wrong...`

Gil80, 2023-04-27:
I find this message to be very confusing.
I think I'm getting really good results:

```
HINT: z position_endstop=2.400 - offset=-0.007500 --> possible z position_endstop=2.407
Z-CALIBRATION: probe=5.420 - (switch=7.380 - nozzle=2.403 + switch_offset=0.450) --> new offset=-0.007500
```

```
[stepper_z]
step_pin: zboard:PB5
dir_pin: !zboard:PB6
enable_pin: !zboard:PC3
endstop_pin: zboard:PA7 # This is the Z-endstop on the Creality board where the sexbolt is connected.
#endstop_pin: probe:z_virtual_endstop
position_endstop: 2.4 ;2.67 # ROUGHLY MEASURE THE POSITION FROM THE TOP OF YOUR BUILD PLATE TO THE TOP OF THE SEXBOLT. SEXBOLT SHOULD BE NO HIGHER THAN 4MM
position_min: -3 # SOME NUMBER BELOW ZERO
```

```
[z_calibration]
nozzle_xy_position: 169,236
#switch_xy_offsets: 145,196
switch_xy_offsets: -24,-42
#bed_xy_position: 117.5,117.5 ; disabled because I'm using bed mesh reference index.
# wiggle_xy_offsets: 0,0
switch_offset: 0.45 #measured 0.41mm #0.45 for textured side, #0.39 for smooth side.
offset_margins: -1.0,1.0
```

Yet, when I run: `CALCULATE_SWITCH_OFFSET` I get: `The resulting switch offset is negative! Either the nozzle is still too far away or something else is wrong...`

Why is that?

TitusLabs, 2023-04-28:
Because the current z position is not in the range to be able to calculate a switch offset. Did you move your nozzle down to zero before? This is step 4 in the documentation: https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calculate_switch_offset

Gil80, 2023-04-28:
@TitusLabs I followed this guide: https://github.com/LoganFraser/VoronMods/tree/main/KlickySetup#determine-your-switch_offset-in-klicky-z-calibrationcfg

Gil80, 2023-04-28:
> Because the current z position is not in the range to be able to calculate a switch offset. Did you move your nozzle down to zero before? This is step 4 in the documentation: https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calculate_switch_offset

Following the guide you kindly provided, this is what I get:
```

switch_offset=0.450 - (current_z=-0.045 - z_offset=-0.045) --> new switch_offset=0.450

CALCULATE_SWITCH_OFFSET

HINT: z position_endstop=2.400 - offset=-0.045000 --> possible z position_endstop=2.445

Z-CALIBRATION: probe=4.610 - (switch=6.625 - nozzle=2.420 + switch_offset=0.450) --> new offset=-0.045000
```

My `switch_offset` was already set to `switch_offset: 0.45`
I used a paper and 0.1mm Z decrements until the nozzle slightly grabbed the piece of paper. On klipperscreen, the Z value was 0.0.

So I still don't get why I see `The resulting switch offset is negative! Either the nozzle is still too far away or something else is wrong...` when I run `CALCULATE_SWITCH_OFFSET` after `CALIBRATE_Z`.
Or do I need to run `CALCULATE_SWITCH_OFFSET` only after I manually get the nozzle to touch the center of the bed?

TitusLabs, 2023-04-28:
Correct, it needs the current z height where you think it should be 0.0. Then it tries to calculate the switch offset based on the first calibration run and this z position.

The command is intended to help a little in determining the switch offset.

In you case, you already seem to have a good starting point from where you can fine tune it by printing some tests :+1:

Gil80, 2023-04-28:
> Correct, it needs the current z height where you think it should be 0.0. Then it tries to calculate the switch offset based on the first calibration run and this z position.
>
> The command is intended [...]

so I ran a test print and then I got:
`Offset is greater than allowed: offset=1.117 > max_deviation=1.000`
so I did another `calibrate_z` followed by `CALCULATE_SWITCH_OFFSET` and I got:
`switch_offset=0.450 - (current_z=0.497 - z_offset=0.397) --> new switch_offset=0.350`
so I set the new value for the switch to be 0.350... I have no idea why this change all of sudden. I didn't change any hardware.

Then a try to print again and I get `Offset is greater than allowed: offset=1.070 > max_deviation=1.000`
I'm lost here...

TitusLabs, 2023-04-28:
Ok, then make some test of your repeatability. And be careful, check z=0 before starting a print. 

And don't get confused by the switch offset and the resulting z offset! You can reduce the z offset by changing the z endstop position. A hint is prompted after every calibration with a recalculated z endstop position based on the determined z offset.

So..
1. change your z endstop position
2. check your z offset by running calibrate_z and go with your nozzle step by step down to 0.0
3. change your switch offset, if needed and check the result again
4. do a test print

Gil80, 2023-04-29:
I measured the endstop position from the top of the sexbolt to the bed surface, and that's 2.67mm.
do you mean I need to lower it or make it taller?

TitusLabs, 2023-04-29:
Did you already have the previous version running or is it your first installation? The update does not change any core functionality.

First of all, configuring the z position endstop is the "old way" of calibrating the z offset. There are many instructions out there for this. The caliper method may not be accurate enough. You need to push the endstop until you just hear the click and no further.

The more correct this value is, the smaller the calibrating offset will be. It's ok to have a resulting offset of more than 1mm - just increase the max_deviation/offset_margins to allow this. But, it's more safe to keep it small. I added the hint for the z position endstop to adapt it in a fast way. It's ok that these values vary - even the resulting z offset varies. That's fine so long the results are correct. 

So, in the beginning, please verify the results before printing!! It is documented here: https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Test-It

If your probe works correctly, calibrate your switch offset first. Test it every time if the nozzle touches the surface with x=0.0. For testing, you can increase your max_deviation. Then calibrate your z position endstop to decrese the resulting z offset. And set a max_deviation that fits your needs.

Gil80, 2023-04-29:
```
Did you already have the previous version running or is it your first installation? The update does not change any core functionality.
```
I had the previous version running and it ran perfectly.

```
First of all, configuring the z position endstop is the "old way" of calibrating the z offset. There are many instructions out there for this. The caliper method may not be accurate enough. You need to push the endstop until you just hear the click and no further.
```
Then I think the documentation should mention the actuation point, i.e., the travel distance until the switch clicks, like in a keyboard. That makes more sense. Or you probably have that in the documentation and I missed it.

As for this guide: https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calculate_switch_offset
I just want to verify if by "Switch Offset" you refer to the travel distance it takes until the actuation point? 

p.s. - I updated my last comment and I broke down the steps where I get confused.

TitusLabs, 2023-04-29:
Hm, the z endstop position is not part of my documentation. It is the offset between the endstop and the print surface to shift the z=0.0 onto the print surface. More on this here: https://www.klipper3d.org/Manual_Level.html

But, there are many things which can change this value for instance the rising chamber temperature. That's why I build this plugin. You can say that it creates an offset from the offset. This is the correct offset of the current state in the system and can be different over time.

To keep the offset of the offset as small as possible, the first offset which is defined by the z endstop position should be calibrated correctly. The output of the z endstop position value is meant as a hint. And because this value is derived from the current offset, it changes too. So, don't worry. With this plugin, the z endstop offset does not have to be perfectly calibrated, but as accurate as possible so that the offset from CALIBRATE_Z is not too large.

The switch offset is a variable in the equation which I cannot probe. It is the distance from the switch body to the trigger point of the switch. That's why this needs to be determined by data sheet or by the new command CALCULATE_SWITCH_OFFSET. In both cases it needs to be fine tuned. This is the variable which you can use to get the first layer exactly as you want. And you need to test it if your chosen switch offset is correct. Set the value in your config, restart Klipper/Firmware and go with z step by step down to 0 and check everytime the left distance to the surface.

I hope this will clear your confusions. And please read again the documentation and let me know if I can clarify anything else..

TitusLabs, 2023-08-21:
I am closing this ticket now. If there is still a problem, please reopen it...

closed 2023-08-21 by TitusLabs

suifong0728, 2023-09-09:
Offset -3.133 is outside the configured range of min=-1.000 and max=1.000
Please help me. My vorontrident 350 is always negative.

TitusLabs, 2023-09-11:
@suifong0728 You can calibrate your z endstop position to keep the calibration offset as small as possible. If you have a V2 with sacking gantry, home your z after QGL... (with offset_margins, you can change the boundaries of -1.0 and 1.0)