protoloft/klipper_z_calibration #91: Nozzle sticks to endstop pin on Sexbolt, pulling it out between nozzle measure and probe measure

Havoc340, 2023-04-27:
Hello. This is not really an issue of the code, but an issue with my build and my inability to program. I can Gcode, but I don't see how to solve this issue with Gcode since it happens during the Z calibration routine itself. 

I'm using a voron/mellow Hartk sexbolt style Z endstop and the pin sticks to the nozzle when I probe hot. On my printer I need to retract Z high enough to clear the Euclid probe, and that distance is enough to pull my endstop pin out if the nozzle sticks to it.  I'm using CHT nozzles and even without filament in the extruder there's enough left in there to ooze during the nozzle measure. 

A "fix" that I would like to see would be a simple X/Y move between the nozzle measure, and the movement to Z clearance height before the probe measure. An "intermediate move" parameter or something like that to break the bond between the pin and nozzle after the last measure move. 

The routine would be kind of like this:
Measure nozzle (like normal)
"Intermediate move" XY (move the nozzle in x y at the user defined amount, likely 2mm in one direction)
Clearance height move Z (The standard clearance height already in the routine, like normal)
Move to probe measure location  (normal)
Measure probe (normal)
do the magic stuff (normal)

TitusLabs, 2023-04-28:
There is now the wiggle_xy_offsets for exactly like this:

wiggle_xy_offsets: 0,0
#   After probing the nozzle and retracting, move x some distance away and
#   back. Useful to prevent the z endstop pin sticking to the nozzle and
#   being pulled out of the assembly. Can be negative. Defaults to zero to
#   disable it. Can be combined in x and y to move diagonally. Be careful
#   to not move your nozzle out of range!

Havoc340, 2023-04-28:
Awesome. I saw "wiggle" in the code when I was looking through it. I thought that might be what I was looking for but I didn't stumble across any documentation on it, but my non-programmer mind couldn't understand exactly what it was supposed to do based off the code itself.  Not to say the documentation isn't available or accessible, rather I have a hard time sometimes finding things, especially if they are right in front of my face haha. 

Thanks for the good work!

Havoc340, 2023-04-28:
Just tried out Wiggle_XY and it is exactly what I needed. Thanks again.

closed 2023-04-28 by Havoc340

TitusLabs, 2023-04-28:
You're welcome, glad you like it!