protoloft/klipper_z_calibration #93: offset_margins  Not effective

ptsa, 2023-04-29:
Offset is greater than allowed: offset=1.210 > max_deviation=1.000

in klicky-z-calibration.cfg i set
offset_margins: -3.0,3.0
But the prompt still exceeds the range

educatedCaveman, 2023-04-29:
I am having a similar issue.  I found that if i added the `max_deviation` line back, it worked.  But I think the code should either be fixed, or the documentation updated.

TitusLabs, 2023-04-29:
For backward compatibility, the max_deviation is used over the offset_margins if it is used in your configuration. The "offset is greater as allowed" is the error message of the max_deviation configuration. So, please check if you still have this in your configuration...

May be I should add a check to not have both defined in the configuration :thinking:

TitusLabs, 2023-05-08:
Sorry, you were right. There was a bug using always the default of max_deviation. This is fixed now in 1.0.1 :+1:

TitusLabs, 2023-05-10:
Is fixed with 1.0.1. If there are still issues, please open a new ticket..

closed 2023-05-10 by TitusLabs