protoloft/klipper_z_calibration #94: "Offset is greater than allowed": leads to probe not docking, crashing the bed

stahlfabrik, 2023-04-29:
Hi,

I upgraded my Voron 2.4 and your plugin to the latest version. Now - I do not why yet, the calibrate_z fails with

"Offset is greater than allowed: offset=1.061 > max_deviation=1.000"

Then the end_gcode seems not to be executed which I think is a bad bug. 
end_gcode:
#   A list of G-Code commands to run after each calibration command.
#   See docs/Command_Templates.md for the G-Code format. This can be used to
#   detach the probe afterwards.
  Dock_Probe
  _SET_Z_CURRENT

Preferably IMHO the print should be canceled - or the end_gcode should be executed.

stahlfabrik, 2023-04-29:
Or is there a way for my print_start gcode to check if calibrate_z was successful and then cancel print on my terms?

stahlfabrik, 2023-04-29:
I have configured
offset_margins: -1.1,1.1

The values seems to get ignored. The third crash just happened:
Offset is greater than allowed: offset=1.069 > max_deviation=1.000

So I have to configure max_deviation still?

This is my current config
`[z_calibration]
nozzle_xy_position: 97,307
#   A X, Y coordinate (e.g. 100,100) of the nozzle, clicking on the z endstop.
switch_xy_position: 93,283
#   A X, Y coordinate (e.g. 100,100) of the probe's switch body, clicking on
#   the z endstop.
#switch_xy_offsets: optional when switch_xy_position is used
#   Instead of an absolute position (switch_xy_position), this configuration
#   adds an X, Y offset (e.g. -6,-18) to the nozzle position.
#bed_xy_position: default from relative_reference_index of bed_mesh
#   A X, Y coordinate (e.g. 100,100) where the print surface (e.g. the center
#   point) is probed. These coordinates are adjusted by the
#   probe's X and Y offsets. The default is the relative_reference_index
#   of the configured bed_mesh, if configured. It's possible to change the relative
#   reference index at runtime or use the GCode argument BED_POSITION of CALIBRATE_Z.
wiggle_xy_offsets: 0,0
#   After probing the nozzle and retracting, move x some distance away and
#   back. Useful to prevent the z endstop pin sticking to the nozzle and
#   being pulled out of the assembly. Can be negative. Defaults to zero to
#   disable it. Can be combined in x and y to move diagonally. Be careful
#   to not move your nozzle out of range!
switch_offset: 0.46
#   The trigger point offset of the used mag-probe switch.
#   A larger value will position the nozzle closer to the bed.
#   This must be determined manually. More on this later
#   in this section..
offset_margins: -1.1,1.1
#   The minimum and maximum margins allowed for the calculated offset.
#   If the offset is outside these values, it will stop!
#   The margin can be defined as "min,max" e.g. "-0.5,1.5" or by just one
#   value e.g. "1.0" which translates to "-1.0,1.0" (which is also the default).
#max_deviation: DEPRECATED - please use offset_margins instead!
#   The maximum allowed deviation of the calculated offset.
#   If the offset exceeds this value, it will stop!
#   The default is 1.0 mm.
#samples: default from "probe:samples" section
#   The number of times to probe each point. The probed z-values
#   will be averaged. The default is from the probe's configuration.
#samples_tolerance: default from "probe:samples_tolerance" section
#   The maximum Z distance (in mm) that a sample may differ from other
#   samples. The default is from the probe's configuration.
#samples_tolerance_retries: default from "probe:samples_tolerance_retries" section
#   The number of times to retry if a sample is found that exceeds
#   samples_tolerance. The default is from the probe's configuration.
#samples_result: default from "probe:samples_result" section
#   The calculation method when sampling more than once - either
#   "median" or "average". The default is from the probe's configuration.
#safe_z_height: 2 * z_offset from the "probe:z_offset" section
#   The absolute z position in mm to move to before moving to the next
#   position. The default is two times the z_offset from the probe's
#   configuration. The minimum safe z height is 3mm.
#clearance: DEPRECATED - please use safe_z_height instead!
#   The distance in mm to move up before moving to the next
#   position. The default is two times the z_offset from the probe's
#   configuration.
#position_min: default from "stepper_z:position_min" section.
#   Minimum valid distance (in mm) used for probing move. The
#   default is from the Z rail configuration.
speed: 200
#   The moving speed in X and Y. The default is 50 mm/s.
#lift_speed: default from "probe:lift_speed" section
#   Speed (in mm/s) of the Z axis when lifting the probe between
#   samples and clearance moves. The default is from the probe's
#   configuration.
#probing_speed: default from "stepper_z:homing_speed" section.
#   The fast probing speed (in mm/s) used, when probing_first_fast
#   is enabled. The default is from the Z rail configuration.
#probing_second_speed: default from "stepper_z:second_homing_speed" section.
#   The slower speed (in mm/s) for probing the recorded samples.
#   The default is second_homing_speed of the Z rail configuration.
#probing_retract_dist: default from "stepper_z:homing_retract_dist" section.
#   Distance to retract (in mm) before probing the next sample.
#   The default is homing_retract_dist from the Z rail configuration.
#probing_first_fast: false
#   If true, the first probing will be faster by the probing speed.
#   This is to get down faster and not record the result as a
#   probing sample. The default is false.
start_gcode:
#   A list of G-Code commands to run before each calibration command.
#   See docs/Command_Templates.md for the G-Code format. This can be used to
#   attach the probe.
  _SET_Z_CURRENT VAL=HOME
  Attach_Probe
#before_switch_gcode:
#   A list of G-Code commands to run before to each probing on the
#   mag-probe. See docs/Command_Templates.md for the G-Code format. This can
#   be used to attach the probe after probing on the nozzle and before probing
#   on the mag-probe.
end_gcode:
#   A list of G-Code commands to run after each calibration command.
#   See docs/Command_Templates.md for the G-Code format. This can be used to
#   detach the probe afterwards.
  Dock_Probe
  _SET_Z_CURRENT`

stahlfabrik, 2023-04-29:
Yup, when configuring max_deviation to 1.1 it works... So that variable is not deprecated really but needed very much?

stahlfabrik, 2023-04-29:
Mainsail updater says I am on version v1.0.0-0-g73d62dc of z_calibration

stahlfabrik, 2023-04-29:
Another (and my last hopefully) input:
Does it cancel the current print start gcode when that error is triggered? Because, IMHO the next command (Dock_Probe_Unlock) should have undocked anyway:
    CALIBRATE_Z
    Dock_Probe_Unlock

So I think CALIBRATE_Z was erroring, then PRINT_START was exited and Klipper started to print with the probe on...

Havoc340, 2023-04-29:
I was having an issue similar to this. My routine WAS like this:

Home printer (xyz)
Quad Gantry Level
Auto Z

It would fail with a 2.3mm error. 

The problem was that my gantry sags a lot. So when I home the printer the first time, it sets the original Z location. Quad Gantry Level moves the gantry and fixes the sag BUT when it does, the nozzle is now over 2mm higher than it used to be. When I go to do AutoZ next, the distance was out of tolerance. 

My fix was to add another home in the routine AFTER QGL, before AutoZ, to reset the Z height with a level gantry. I also did a manual XY to my pin location to be sure not to drag the euclid probe across the bed into the pin. So it now looks like this:

    BED_MESH_CLEAR
    G28                                                  # initial home 
    M117 Quad Gantry Level...              #screen message to tell me what it's doing next  
    DEPLOY_PROBE                               #deploy euclid probe
    G0 Z20                                            # safe move to clear stuff
    G0 Y300                                          # safe move to clear stuff
    QUAD_GANTRY_LEVEL_ORIGINIAL
    M117 Calibrate Z                            #screen message to tell me what it's doing next  
    G0 X341.6 Y419.5                           #safe move to put nozzle above pin again
    G28 Z                                             #reset Z home after QGL
    CALIBRATE_Z                                 #calibrate Z
    M117 Generate Mesh                   #screen message to tell me what it's doing next  
    BED_MESH_CALIBRATE                 #run mesh
    STOW_PROBE                                #put probe away
    
    
    I will say that my config is custom built by me and I don't know exactly what I'm doing, but I go this to work. Also my routines are a little modified to keep the probe deployed during QGL, Autoz and the bed mesh, so I only deploy and stow my probe once. 
    
   I do not know if this is your problem you are having, but I had the same error and homing Z after QGL but before calibrate_z is how I fixed it.

TitusLabs, 2023-04-29:
I'm sorry or your crash!

But yes, a z homing after QGL helps to keep the offset small. I should have mentioned this in my documentation.

So, if I understand you correctly, the end gcode is not executed, the error is printed and the printer continues to print?

I can't quite understand that. At this point the code is as it has always been. If the offset is out of the configured range, the end gcode is executed and an error is raised. This should stop the print and it always have in my case :thinking: 

Is your clearance/safe_z_height configured correctly? Or what exactly is your move for?

stahlfabrik, 2023-04-30:
I do a Z homing after QGL. Mechanically, the thing that is different now than when I set up auto calibrate z plugin is that I use a different PEI sheet - so I guess there is the root cause for me getting now a different offset. 

So yes, the end gcode is not executed, when the offset is bigger than the max_deviation. So the klicky probe is not getting docked and the print starts immediately - I even think that the rest of my start_print gcode is also not executed. Because I think in the end Dock_Probe_Unlock should have docked the probe as well, right?

So if you want to experience my issue - I would think you would have to configure max_deviation to be smaller than the offset and see thing go downhill:-)

Another point I wanted to underline: if I do not configure max_deviation to be 1.1 (but comment max_deviation out completely, your plugin thinks it is 1.0 - even though I configured the offset_margins: -1.1,1.1 
I read the available information as if the offset_margins would be used, if max_deviation is not configured. And max_deviation to be obsolete. But to me it seems that this ist not working as expected.

As my print_start macro is a little fancy I did not want to post it here. But I do now, so you can understand better what I do - at least the relevant part. 

```
 {% elif state == 'perform_remaining_actions' %}
    RESUME_BASE
    SET_GCODE_VARIABLE MACRO=PRINT_START VARIABLE=state VALUE='"initialize"'
    _LCD_KNOB COLOR=RED
    # Heat up and wait for extruder
    M117 Heat extruder
    {action_respond_info("Heat extruder")}
    M109 S{ps.extruder_target}
    # Turn off part cooling fan
    M107

    Attach_Probe_Lock
    M117 QGLing...
    {action_respond_info("Quad gantry leveling...")}
    QUAD_GANTRY_LEVEL
    M117 Homing...
    {action_respond_info("Homing...")}
    G28 Z

    M117 Bed mesh leveling...
    {action_respond_info("Bed mesh leveling...")}
    BED_MESH_CALIBRATE
    CLEAN_NOZZLE_NO_PURGE
    CALIBRATE_Z
    Dock_Probe_Unlock

    SET_GCODE_OFFSET Z_ADJUST={ps.z_adjust} MOVE=1

    # Set target temp for exhaust fan
    M141 S{ps.chamber_target}

    STATUS_PRINTING
    M117 Prime Line...
    {action_respond_info("Prime Line...")}
    LINE_PURGE
    M117 Printing...
    {action_respond_info("Printing...")}
    G92 E0
    # Metric values
    G21
    # Absolute positioning
    G90 
    # set extruder to relative mode
    M83
    UPDATE_DELAYED_GCODE ID=_CLEAR_DISPLAY DURATION=1
  {% endif %}
```

TitusLabs, 2023-05-03:
I have just tested it again and it docks the probe using the end_gcode, raises the error and stops the print.

Your PRINT_START macro is really complicated. I don't know if this would cause the problem. I cannot do more than throw an error. But never the less, it should execute the end_gcode before raising the error!

TitusLabs, 2023-05-16:
I have released a bugfix which should eliminate the max_deviation error if configured properly. But, you should find out why your print does not stop...

closed 2023-06-27 by TitusLabs