protoloft/klipper_z_calibration #95: switch hitting the z switch

Amoulier, 2023-04-30:
I updated the z_calibration today and now each time the printer try to **z_calibrate** the nozzle the switch get disconected because when the head move the klikly probe hit the "z switch" disconecting it.

I have tried to find where can i make the head move up a little more so it do not hit the z switch but was unable to find it.

TitusLabs, 2023-05-01:
This is strange, there should be no difference after updating to the new version. Although, there is a change in this behavior ...but, you need to switch from the clearance to the safe_z_height property. How is your configuration?

And that's already the answer, set the safe_z_height property

Amoulier, 2023-05-04:
well I solved.... "If something is not broken why updated?" I just whent back to the previous version and its working fine!

itssunnyb, 2023-05-07:
Having the same issue. What has changed? I updated today now my switch is hitting the z switch. I don't have a setting for Safe_Z_height. I have a setting for Variable_z_height but does nothing. How do I install the old version I have auto update set for through moonraker.

TitusLabs, 2023-05-08:
So, you had a working system with the auto calibration and updated it to the current version.. and then? When exactly does it hit the z endstop? On which move/step? How does your configuration look like?

The safe_z_height is a new property. The old one was the clearance and does still work. Do you have the clearance configured? And what value is your z_offset under [probe]?

I would like to understand why your update process does not work. And to solve this, don't downgrade, set a good safe_z_height :+1:

bithoarder, 2023-06-04:
I have a simular problem, to me it looks like safe_z_height is not used between attaching the probe and clicking with the switch body.
I have this quick fix:
```
[z_calibration]
nozzle_xy_position:    208.0, 309.0
switch_xy_offsets:     4, -25 # 212, 284
bed_xy_position:       175.0, 150.0
safe_z_height:         20
...
before_switch_gcode:  ATTACH_PROBE_FOR_Z_CALIBRATION

[gcode_macro ATTACH_PROBE_FOR_Z_CALIBRATION]
gcode:
    ATTACH_PROBE
    G0 Z20
```

TitusLabs, 2023-06-27:
Hm, I would not exclude the possibility that I have overlooked something here. If I look into the code, the safe_z_height is always used when a probing procedure is started. But, I do a move up in my attach macro too. I will test it without it...

TitusLabs, 2023-06-28:
This is weird. I just double checked the safe_z_height with start_gcode and before_switch_gcode and removed the move up in my attach macro. And it's always moving up to safe_z_height first before moving to the probing position.

Soo, the only thing I can think about is maybe your `G0 Z20` is a relative move while the safe_z_height is an absolute value and is not enough for your system? :thinking:

Amoulier, 2023-07-07:
> So, you had a working system with the auto calibration and updated it to the current version.. and then? When exactly does it hit the z endstop? On which move/step? How does your configuration look [...]

well I gave it another try, today and just fixed the "z_offset: " setting in my probe section on the printer.cfg and it worket like before!

TitusLabs, 2023-07-08:
OK, yes! If you don't use the safe_z_height or the old clearance setting, then the z_offset of the probe will be used :+1: