protoloft/klipper_z_calibration #97: Z Ofsset isue to park my Euclid

canadank0, 2023-05-16:
Hello,
I dont know if i misread something, but when your script apply the Z offset it also change Z offset number for returning my euclid probe to plastic pocket. In my case it is lower number and the probe cannot find the entry and fall down. is there way to apply the offset after probe is parked and the nozzle is over the bed?

TitusLabs, 2023-05-16:
Yes, maybe I missed that with the order :thinking: It's historically grown. But, it would just shift your problem to your next probing attempt. The solution is (I think I should document this) to set the offset to 0 in your (un)dock macros if you have it mounted to the bed like this:

```
SAVE_GCODE_STATE NAME=YOUR_MACRO_NAME
SET_GCODE_OFFSET Z=0.0

your macro content here...

RESTORE_GCODE_STATE NAME=YOUR_MACRO_NAME
```

So, you save the current offset, set it back to 0.0 with which you defined your dock offsets, dock or undock your probe and restore the offset.

closed 2023-05-17 by canadank0

devsfan1830, 2023-08-04:
I think documentation of this issue and solution is needed. I started having it with my Euclid as well. I can dock/undock fine for QGL and Mesh, but when I do the auto-z, as it dock it moves in significantly lower and causes it to miss. MOST of the time the switch just falls behind the bed. Once in a while it gets stuck dangling funny from the toolhead. I'm gonna try your suggested solution but wanted to clarify. I'd do this in BOTH dock and undock macros?