protoloft/klipper_z_calibration #98: offset outside of configured range?

Gil80, 2023-05-20:
Looking at the Z Calibration output, I thought I got it bang on, but when I send it to print, I get the following error: `Offset -1.025 is outside the configured range of min=-1.000 and max=1.000`

`HINT: z position_endstop=1.230 - offset=0.000000 --> possible z position_endstop=1.230
Z-CALIBRATION: probe=6.395 - (switch=7.265 - nozzle=1.260 + switch_offset=0.390) --> new offset=0.000000`

Why is that?

TitusLabs, 2023-05-21:
It's so hard to say.  Maybe you had plastic at the nozzle, your gantry sacks and you did not home after QGL, or there is something else wrong with your mechanics/klicky...

RickDnamps, 2023-05-26:
I don't know why but they change the code now and the maximum offset is not adjustable anymore even if you but somehting for the variable like = 2.00 .... the code is hardcoded with min max offset = -1.00 ... +1.00. That is sad cause now if you just want to change you print plat for another in another material Carbon fiber for me with a thicker thickness I have to change manually my  position_endstop in my print.cfg to add or remove 1.00mm. 

So if in your case if you want it to work I suggest Redo  your Z adjustment direct on the printer with paper and save the value... or just add 1.00 to your current value "position_endstop" at the end of the print.cfg (will be in green probably)  like that: 
#*# [stepper_z]
#*# position_endstop = 1.443

So in your case the new value will become 
#*# position_endstop = 2.443

Use your own value not mine ...

TitusLabs, 2023-05-27:
@RickDnamps are you using the latest version? There was a bug with the default value of the max_deviation. But I suggest to use the newer property: offset_margins

RickDnamps, 2023-05-27:
> @RickDnamps are you using the latest version? There was a bug with the default value of the max_deviation. But I suggest to use the newer property: offset_margins

I just Saw that the update need to change some parameter in Z_Calibration.cfg .... Probably alot of people do like me and just update it without changing anything to their config file .... 

I will update that now and retry the latest version thx

TitusLabs, 2023-05-27:
Exactly, there have been deprecated configurations for over a year that I have now removed. The max_deviation is now only marked as deprecated with the current release, so that the update can be completed without any adjustments if possible (if you were almost up to date). So there is no need to switch to offset_margins right now - but it would be great.

TitusLabs, 2023-08-21:
I am closing this ticket now. If there is still a problem, please reopen it...

closed 2023-08-21 by TitusLabs