protoloft/klipper_z_calibration #99: Slight Change in z_calibration.cfg

AustinTylerDean, 2023-05-25:
I believe you need to remove a space from "safe_z_height: 2 * z_offset from "probe:z_offset" section"

The space is right after 2, so should be "2*"

Otherwise I am getting a parsing error, though I haven't even achieved flight on my 2.4...

AustinTylerDean, 2023-05-25:
When the value from the initial unconfigured z_calibration.cfg reads "default from "xxx" section, am I being dumb for thinking the parser knows what to do with that?

AustinTylerDean, 2023-05-25:
Unless there's a value, you need to # the line or put a value in.

closed 2023-05-25 by AustinTylerDean

TitusLabs, 2023-05-25:
Correct, it's the default value the script uses if you do not configure a value for this property :+1: 
But, you're right, I forgot the "default" there..