protoloft/klipper_z_calibration #101: bed mesh not working

kingbusa, 2023-05-27:
[bed_mesh hot]
#*# version = 1
#*# points =
#*# 	  0.060000, 0.037500, 0.022500, 0.011250, 0.025000, 0.012500, 0.035000
#*# 	  0.018750, 0.016250, -0.007500, -0.006250, 0.011250, 0.027500, 0.043750
#*# 	  -0.016250, -0.023750, -0.020000, -0.016250, -0.010000, 0.013750, 0.032500
#*# 	  -0.046250, -0.057500, -0.050000, -0.006250, -0.031250, -0.018750, -0.010000
#*# 	  -0.040000, -0.045000, -0.048750, -0.042500, -0.036250, -0.016250, -0.020000
#*# 	  0.018750, -0.003750, 0.000000, -0.002500, 0.010000, 0.012500, 0.002500
#*# 	  0.097500, 0.063750, 0.061250, 0.063750, 0.073750, 0.07375   old bed mesh
[bed_mesh hot]
#*# version = 1
#*# points =
#*# 	  4.904448, 4.878198, 4.849448, 4.840698, 4.853198, 4.863198, 4.894448
#*# 	  4.888198, 4.875698, 4.846948, 4.844448, 4.858198, 4.884448, 4.918198
#*# 	  4.868198, 4.851948, 4.844448, 4.841948, 4.849448, 4.878198, 4.904448
#*# 	  4.834448, 4.814448, 4.814448, 4.848198, 4.828198, 4.845698, 4.864448
#*# 	  4.838198, 4.828198, 4.816948, 4.816948, 4.820698, 4.840698, 4.848198
#*# 	  4.878198, 4.846948, 4.846948, 4.841948, 4.850698, 4.859448, 4.861948
#*# 	  4.931948, 4.895698, 4.886948, 4.888198, 4.900698, 4.906948, 4.908198 new bed mesh 

i did a reinstall and still it is not getting better any help would be great thanks

TitusLabs, 2023-05-28:
I think your mesh is configured without a relative reference index :thinking: 

If you configure a relative reference index for your mesh, this point becomes 0.0 and every other point becomes relative to this point. And this point/position must be probed with the z calibration. You can achieve this by not configuring a bed_xy_position.

What is the output of your z calibration?

TitusLabs, 2023-08-21:
I am closing this ticket now. If there is still a problem, please reopen it...

closed 2023-08-21 by TitusLabs