protoloft/klipper_z_calibration #103: Manual z offset distance?

BuildingOutLoud, 2023-06-17:
Hi, I have an i3 style printer where I can't quite get the nozzle to touch a z endstop outside the bed area but I can touch another part of the toolhead to the z endstop. I know the z offset from that part to the nozzle, and can CALIBRATE_Z successfully, although the resulting offset is off by the previously mentioned z distance from the toolhead part to the nozzle.

Is there a way to manually add an offset to the calculation? or am I way off in how this all works?

TitusLabs, 2023-06-27:
Hm, the great thing is that it works like that in every situation. Even if your toolhead gets warped somehow or you use a different nozzle. In principle, you can still make a `SET_GCODE_OFFSET Z_ADJUST=+-some_value` AFTER the calibration. This is documented here for textured plates for instance: https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calibrate_z

But there are also solutions where the end stop is swiveled in with a servo...

BuildingOutLoud, 2023-07-09:
I hadn't thought of that. Thanks! :D

closed 2023-07-09 by BuildingOutLoud