protoloft/klipper_z_calibration #104: Relative Reference Index Deprecated

AustinTylerDean, 2023-06-21:
Is the deprecation going to interfere with this plugin/config?

todstelzer, 2023-06-21:
It´s broken...

todstelzer, 2023-06-21:
Got it working again.

`# Klipper plugin for a self-calibrating Z offset.
#
# Copyright (C) 2021-2023  Titus Meyer <info@protoloft.org>
#
# This file may be distributed under the terms of the GNU GPLv3 license.
import logging
from mcu import MCU_endstop

class ZCalibrationHelper:
    def __init__(self, config):
        self.state = None
        self.z_endstop = None
        self.z_homing = None
        self.last_state = False
        self.last_z_offset = 0.
        self.position_z_endstop = None
        self.config = config
        self.printer = config.get_printer()
        self.switch_offset = config.getfloat('switch_offset', 0.0, above=0.)
        # max_deviation is deprecated
        self.max_deviation = config.getfloat('max_deviation', None, above=0.)
        self.offset_margins = self._get_offset_margins('offset_margins',
                                                     '-1.0,1.0')
        self.speed = config.getfloat('speed', 50.0, above=0.)
        # clearance is deprecated
        self.clearance = config.getfloat('clearance', None, above=0.)
        self.safe_z_height = config.getfloat('safe_z_height', None, above=0.)
        self.samples = config.getint('samples', None, minval=1)
        self.tolerance = config.getfloat('samples_tolerance', None, above=0.)
        self.retries = config.getint('samples_tolerance_retries',
                                     None, minval=0)
        atypes = {'none': None, 'median': 'median', 'average': 'average'}
        self.samples_result = config.getchoice('samples_result', atypes,
                                               'none')
        self.lift_speed = config.getfloat('lift_speed', None, above=0.)
        self.probing_speed = config.getfloat('probing_speed', None, above=0.)
        self.second_speed = config.getfloat('probing_second_speed',
                                            None, above=0.)
        self.retract_dist = config.getfloat('probing_retract_dist',
                                            None, above=0.)
        self.position_min = config.getfloat('position_min', None)
        self.first_fast = config.getboolean('probing_first_fast', False)
        self.nozzle_site = self._get_xy("nozzle_xy_position", True)
        self.switch_site = self._get_xy("switch_xy_position", True)
        self.switch_xy_offsets = self._get_xy("switch_xy_offsets", True)
        self.bed_site = self._get_xy("bed_xy_position", True)
        self.wiggle_offsets = self._get_xy("wiggle_xy_offsets", True)
        gcode_macro = self.printer.load_object(config, 'gcode_macro')
        self.start_gcode = gcode_macro.load_template(config, 'start_gcode', '')
        self.switch_gcode = gcode_macro.load_template(config,
                                                      'before_switch_gcode',
                                                      '')
        self.end_gcode = gcode_macro.load_template(config, 'end_gcode', '')
        self.query_endstops = self.printer.load_object(config,
                                                       'query_endstops')
        self.printer.register_event_handler("klippy:connect",
                                            self.handle_connect)
        self.printer.register_event_handler("homing:home_rails_end",
                                            self.handle_home_rails_end)
        self.gcode = self.printer.lookup_object('gcode')
        self.gcode.register_command('CALIBRATE_Z', self.cmd_CALIBRATE_Z,
                                    desc=self.cmd_CALIBRATE_Z_help)
        self.gcode.register_command('PROBE_Z_ACCURACY',
                                    self.cmd_PROBE_Z_ACCURACY,
                                    desc=self.cmd_PROBE_Z_ACCURACY_help)
        self.gcode.register_command('CALCULATE_SWITCH_OFFSET',
                                    self.cmd_CALCULATE_SWITCH_OFFSET,
                                    desc=self.cmd_CALCULATE_SWITCH_OFFSET_help)
[... 370 lines cut ...]
        if (self.max_deviation is not None # deprecated
            and abs(offset) > self.max_deviation):
            self.helper.end_gcode.run_gcode_from_command()
            raise self.helper.printer.command_error("Offset is greater than"
                                                    " allowed: offset=%.3f"
                                                    " > max_deviation=%.3f"
                                                    % (offset,
                                                    self.max_deviation))
        elif (offset < self.offset_margins[0]
              or offset > self.offset_margins[1]):
            self.helper.end_gcode.run_gcode_from_command()
            raise self.helper.printer.command_error("Offset %.3f is outside"
                                                    " the configured range of"
                                                    " min=%.3f and max=%.3f"
                                                    % (offset,
                                                       self.offset_margins[0],
                                                       self.offset_margins[1]))
        # set new offset
        self._set_new_gcode_offset(offset)
        # set states
        self.helper.last_state = True
        self.helper.last_z_offset = offset
        self.helper.end_gcode.run_gcode_from_command()
def load_config(config):
    return ZCalibrationHelper(config)`

todstelzer, 2023-06-21:
Got it working.
[z_calibration.zip](https://github.com/protoloft/klipper_z_calibration/files/11824281/z_calibration.zip)

todstelzer, 2023-06-21:
Seems to work as well now. Praise chatGPT for it...

`
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BED_MESH_CALIBRATE_BASE
; gcode parameters
variable_parameter_AREA_START : 0,0
variable_parameter_AREA_END : 0,0
; the clearance between print area and probe area 
variable_mesh_area_offset : 5.0
; number of sample per probe point
variable_probe_samples : 2
; minimum probe count
variable_min_probe_count : 3
; scale up the probe count, should be 1.0 ~ < variable_max_probe_count/variable_min_probe_count
variable_probe_count_scale_factor : 1.0
; enable preference index
variable_zero_reference_position : True
gcode:
    ATTACH_PROBE
    {% if params.AREA_START and params.AREA_END %}
        {% set bedMeshConfig = printer["configfile"].config["bed_mesh"] %}
        {% set safe_min_x = bedMeshConfig.mesh_min.split(",")[0]|float %}
        {% set safe_min_y = bedMeshConfig.mesh_min.split(",")[1]|float %}
        {% set safe_max_x = bedMeshConfig.mesh_max.split(",")[0]|float %}
        {% set safe_max_y = bedMeshConfig.mesh_max.split(",")[1]|float %}

        {% set area_min_x = params.AREA_START.split(",")[0]|float %}
	{% set area_min_y = params.AREA_START.split(",")[1]|float %}
	{% set area_max_x = params.AREA_END.split(",")[0]|float %}
	{% set area_max_y = params.AREA_END.split(",")[1]|float %}

	{% if bedMeshConfig.probe_count.split(",")|length == 2 %}
            {% set meshPointX = bedMeshConfig.probe_count.split(",")[0]|int %}
            {% set meshPointY = bedMeshConfig.probe_count.split(",")[1]|int %}
        {% else %}
            {% set meshPointX = bedMeshConfig.probe_count.split(",")[0]|int %}
            {% set meshPointY = bedMeshConfig.probe_count.split(",")[0]|int %}
        {% endif %}

	{% set meshMaxPointX = meshPointX %}
	{% set meshMaxPointY = meshPointY %}

        {% if (area_min_x < area_max_x) and (area_min_y < area_max_y) %}
            {% if area_min_x - mesh_area_offset >=  safe_min_x %}
                {% set area_min_x = area_min_x - mesh_area_offset %}
            {% else %}
                {% set area_min_x = safe_min_x %}
            {% endif %}

            {% if area_min_y - mesh_area_offset >=  safe_min_y %}
                {% set area_min_y = area_min_y - mesh_area_offset %}
            {% else %}
                {% set area_min_y = safe_min_y %}
            {% endif %}

            {% if area_max_x + mesh_area_offset <=  safe_max_x %}
                {% set area_max_x = area_max_x + mesh_area_offset %}
            {% else %}
                {% set area_max_x = safe_max_x %}
            {% endif %}

            {% if area_max_y + mesh_area_offset <=  safe_max_y %}
                {% set area_max_y = area_max_y + mesh_area_offset %}
            {% else %}
                {% set area_max_y = safe_max_y %}
            {% endif %}

            {% set meshPointX = (meshPointX * (area_max_x - area_min_x) / (safe_max_x - safe_min_x) * probe_count_scale_factor)|round(0)|int %}
            {% if meshPointX < min_probe_count %}
                {% set meshPointX = min_probe_count %}
            {% endif %}
	    {% if meshPointX > meshMaxPointX %}
                {% set meshPointX = meshMaxPointX %}
            {% endif %}

            {% set meshPointY = (meshPointY * (area_max_y -area_min_y ) / (safe_max_y - safe_min_y) * probe_count_scale_factor )|round(0)|int %}
            {% if meshPointY < min_probe_count %}
                {% set meshPointY = min_probe_count %}
            {% endif %}
	    {% if meshPointY > meshMaxPointY %}
                {% set meshPointY = meshMaxPointY %}
            {% endif %}

            {% set algorithm = "bicubic" %}
	    {% if "algorithm" in bedMeshConfig %}
	        {% set algorithm = bedMeshConfig.algorithm %}
            {% endif %}
            {% if meshPointX >=7 or meshPointY >=7 %}
                {% set algorithm = "bicubic" %}
            {% endif %}

            {% if zero_reference_position %}
                {% set relative_reference_index = (meshPointX * meshPointY / 2 - 1 )|round(0)|int %}
                BED_MESH_CALIBRATE_BASE mesh_min={area_min_x},{area_min_y} mesh_max={area_max_x},{area_max_y} probe_count={meshPointX},{meshPointY} samples={probe_samples|int} algorithm={algorithm} zero_reference_position={area_max_x},{area_max_y}
            {% else %}
                BED_MESH_CALIBRATE_BASE mesh_min={area_min_x},{area_min_y} mesh_max={area_max_x},{area_max_y} probe_count={meshPointX},{meshPointY} samples={probe_samples|int} algorithm={algorithm}
            {% endif %}
        {% else %}
            BED_MESH_CALIBRATE_BASE
        {% endif %}
    {% else %}
        BED_MESH_CALIBRATE_BASE
    {% endif %}
    DOCK_PROBE
`

AustinTylerDean, 2023-06-21:
Did you use ChatGPT to fix this? If so, I would LOVE the transcript!

Secondly, I hope it helped Protoloft, if it indeed fixed the problem.

todstelzer, 2023-06-21:
Its in German. I had to poke him multiple times to get what i wanted :)

For the z_calibration.py it was trial and error till i found out that i have to use printer.zero_reference_position to get the new zero_reference_position value into the *.py file.
No chatGPT here.

For the variable bed mesh macro i used ChatGPT.
I didnt know how to pares x and y to a single parameter and he helped me there.
zero_reference_position={area_max_x},{area_max_y}

I Just printed a test part and it seems to work ^.^

AustinTylerDean, 2023-06-21:
Awesome! Thank you very much! I hope this is all helpful to the author, as future releases will definitely cause people to come here asking the same question.
Appreciated, Todstelzer!

todstelzer, 2023-06-21:
Ok i´m not sure whats going on but it stopped working....
I tested it and did print a small part but when i tried second part nothing worked again.

todstelzer, 2023-06-22:
Ok have to go to sleep now but i figured it out so i can use the printer at least.
I can´t get the new value zero_reference_position to applied to self.bed_site inside z_calibration.
I set it now to 175,175. Its the middle of my bed and should work on smaller beds as well.

The new bed mesh dont use a index value. Instead, as far i can tell, it probes the zero_reference_position as the last point in the mesh. So the probe point is now fixed and not tied to the bed mesh anymore.
![grafik](https://github.com/protoloft/klipper_z_calibration/ass
[z_calibration.zip](https://github.com/protoloft/klipper_z_calibration/files/11827367/z_calibration.zip)
ets/67606371/53755b4e-644a-4f9e-8a30-041eaa5906b5)

todstelzer, 2023-06-24:
You can add bed_xy_position: xxx, yyy to z_calibration.cfg and set the zero_reference_position there. 
It must be the same value as in printer.cfg [bed_mesh] zero_reference_position value

yanfoo, 2023-06-26:
> You can add bed_xy_position: xxx, yyy to z_calibration.cfg and set the zero_reference_position there. It must be the same value as in printer.cfg [bed_mesh] zero_reference_position value

Thanks,
Your py file with bed_xy_position and zero_reference_position value fixed it for me.

TitusLabs, 2023-06-26:
Sorry, I was just prevented for quite a while. I will fix this tomorrow...

todstelzer, 2023-06-27:
> You can add bed_xy_position: xxx, yyy to z_calibration.cfg and set the zero_reference_position there. It must be the same value as in printer.cfg [bed_mesh] zero_reference_position value

It should work adding bed_xy_position: xxx, yyy  to z_calibration or using the modded py file.

yanfoo, 2023-06-27:
This is exactly what I did and used winscp to do it.
The out of range is something else, it should say what the measured range is and compare to what you set you may need to increase the range.

VORONENTHUSIAST, 2023-06-27:
> This is exactly what I did and used winscp to do it. The out of range is something else, it should say what the measured range is and compare to what you set you may need to increase the range. Sent from Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

Yeah, it's really strange, it's putting X to 360 something, and y to around 680?! I'm on a 310x310 bed so tried 155 x and 155 y as well as 162.5 for the bed_xy_position (came from my old klippy log when using the relative_reference_index before updating) yet both produced this result, I have given up for now, as removing the z_calibrate from my start macro has gotten me printing with an adaptive mesh, but it would be nice to have again!

todstelzer, 2023-06-27:
Can you post/link your printer.cfg [bed_mesh] section and calibrate_z.cfg?
Maybe its a typo somewhere.

VORONENTHUSIAST, 2023-06-27:
> Can you post/link your printer.cfg [bed_mesh] section and calibrate_z.cfg? Maybe its a typo somewhere.
> 
> nozzle_xy_position: 227,353
> 
> Maybe you used a . instead of a , ?

Thank you for being so willing to take a look for me :)

This is my bed_mesh section: 

[bed_mesh]
speed: 175
horizontal_move_z: 10
##--------------------------------------------------------------------
##	Uncomment below for 250mm build
#mesh_min: 40, 40
#mesh_max: 210, 210

##	Uncomment for 300mm build
#mesh_min: 40, 40
#mesh_max: 275, 275

##	Uncomment for 350mm build
#mesh_min: 40, 40
#mesh_max: 310, 310

## Uncomment for Custom build plate Size
mesh_min: 40, 40
mesh_max: 285, 285
##--------------------------------------------------------------------
zero_reference_position: 155, 155
probe_count: 7,9                    # Values should be odd, so one point is directly at bed center
mesh_pps: 2, 3
fade_start: 1
fade_end: 10
algorithm: bicubic
bicubic_tension: 0.2

Here is my calibrate_z:

[z_calibration]
nozzle_xy_position: 196.5, 310
switch_xy_offsets: 5.5, -19.43
bed_xy_position: 155, 155
wiggle_xy_offsets: 185, 310
switch_offset: 3.44
safe_z_height: 5
#position_min: 0.625  (Hashed out as reading directly from printer.cfg after using SAVE_CONFIG command)
start_gcode: Attach_Probe
#before_switch_gcode:
end_gcode: Dock_Probe

Sorry for not attaching a link for them, not 100% sure how to do it (I know, I'm sounding really stupid right now!)

VORONENTHUSIAST, 2023-06-27:
Oh, if needed, here is my bed_mesh_calibrate macro - I have gone for KAMP:

[gcode_macro SETUP_KAMP_MESHING]
gcode:
  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=display_parameters   VALUE={params.DISPLAY_PARAMETERS|default(True)|int}
  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=led_enable   VALUE={params.LED_ENABLE|default(False)|int}
  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=status_macro VALUE='"{params.STATUS_MACRO|default('status_meshing')|string}"'

  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=fuzz_enable VALUE={params.FUZZ_ENABLE|default(False)|int}
  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=fuzz_min    VALUE={params.FUZZ_MIN|default(0)|float}
  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=fuzz_max    VALUE={params.FUZZ_MAX|default(4)|float}

  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=probe_dock_enable  VALUE={params.PROBE_DOCK_ENABLE|default(False)|int}
  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=attach_macro VALUE='"{params.ATTACH_MACRO|default('Attach_Probe')|string}"'
  SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=detach_macro VALUE='"{params.DETACH_MACRO|default('Dock_Probe')|string}"'

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: _BED_MESH_CALIBRATE

### This section configures mesh point fuzzing, which allows probe points to be varied slightly if printing multiples of the same G-code file.

variable_fuzz_enable: False           # Enables/disables the use of mesh point fuzzing to slightly randomize probing points to spread out wear on a build surface, default is False.
variable_fuzz_min: 2                  # If enabled, the minimum amount in mm a probe point can be randomized, default is 0.
variable_fuzz_max: 7                  # If enabled, the maximum amount in mm a probe point can be randomized, default is 4.

### This section is for configuring a mesh margin, which allows the probed mesh to be expanded outwards from the print area.

variable_margin_enable: False           # Enables/disables adding a margin to the meshed area to pad a mesh out for specific needs, default is False.
variable_margin_size: 5                # Size in millimeters to expand the mesh outwards from the print area in all directions.

### This section is for those using a dockable probe that is stored outside of the print area. ###

variable_probe_dock_enable: False       # Enables/disables the use of a dockable probe that is stored outside of the print area, default is False.
variable_attach_macro: 'Attach_Probe'   # Here is where you define the macro that ATTACHES the probe to the printhead. E.g. 'Attach_Probe'
variable_detach_macro: 'Dock_Probe'     # Here is where you define the macro that DETACHES the probe from the printhead. E.g. 'Dock_Probe'

### This section is for those who are using Moonraker's Update Manager for KAMP, or want a more verbose macro. ###

variable_display_parameters: True       # Display macro paramters in the console, useful for debugging the SETUP_KAMP_MESHING call, or more verbosity.

gcode:

    {% if display_parameters == True %}
      { action_respond_info("status_macro: \'%s\'" % (status_macro))  }
      { action_respond_info("fuzz_enable : %d" % (fuzz_enable))  }
      { action_respond_info("fuzz_min    : %f" % (fuzz_min))  }
      { action_respond_info("fuzz_max    : %f" % (fuzz_max))  }
      { action_respond_info("probe_dock_enable: %d" % (probe_dock_enable))  }
      { action_respond_info("attach_macro: \'%s\'" % (attach_macro))  }
      { action_respond_info("detach_macro: \'%s\'" % (detach_macro))  }
    {% endif %}
    
    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}
    {% set bed_mesh_min = printer.configfile.settings.bed_mesh.mesh_min %}
    {% set bed_mesh_max = printer.configfile.settings.bed_mesh.mesh_max %}
    {% set probe_count = printer.configfile.settings.bed_mesh.probe_count %}
    {% set probe_count = probe_count if probe_count|length > 1 else probe_count * 2  %}
    {% set max_probe_point_distance_x = ( bed_mesh_max[0] - bed_mesh_min[0] ) / (probe_count[0] - 1)  %}
    {% set max_probe_point_distance_y = ( bed_mesh_max[1] - bed_mesh_min[1] ) / (probe_count[1] - 1)  %}
    {% set x_min = all_points | map(attribute=0) | min | default(bed_mesh_min[0]) %}
    {% set y_min = all_points | map(attribute=1) | min | default(bed_mesh_min[1]) %}
    {% set x_max = all_points | map(attribute=0) | max | default(bed_mesh_max[0]) %}
    {% set y_max = all_points | map(attribute=1) | max | default(bed_mesh_max[1]) %}
    
    {% if margin_enable == False %}
        {% set margin_size = 0 %}
    {% endif %}
    
    { action_respond_info("{} object points, clamping to bed mesh [{!r} {!r}]".format(
        all_points | count,
[... 33 lines cut ...]

    {% set points_x = [points_x, min_points]|max  %}
    {% set points_y = [points_y, min_points]|max  %}
    { action_respond_info( "Points: x: {}, y: {}".format(points_x, points_y) ) }

    {% if printer.configfile.settings.bed_mesh.relative_reference_index is defined %}
        {% set ref_index = (points_x * points_y / 2) | int %}
        { action_respond_info( "Reference index: {}".format(ref_index) ) }
    {% else %}
        {% set ref_index = -1 %}
    {% endif %}

    {% if probe_dock_enable == True %}
        {attach_macro}              # Attach/deploy a probe if the probe is stored somewhere outside of the print area
    {% endif %}

    {% if led_enable == True %}
        {status_macro}              # Set status LEDs
    {% endif %}

    _BED_MESH_CALIBRATE mesh_min={x_min},{y_min} mesh_max={x_max},{y_max} ALGORITHM={algorithm} PROBE_COUNT={points_x},{points_y} RELATIVE_REFERENCE_INDEX={ref_index}

    {% if probe_dock_enable == True %}
        {detach_macro}              # Detach/stow a probe if the probe is stored somewhere outside of the print area
    {% endif %}

todstelzer, 2023-06-27:
Hmm looks good.
Did you try and switch on/off the printer already?
And what happens when you use my .py with commented out bed_xy_position: 155, 155?
In the .py the position is 175,175 if there is no bed_xy_position defined.

If thats not fix it you have to wait for @TitusLabs to change the code.

VORONENTHUSIAST, 2023-06-27:
I did the switch off and on, sadly no dice, I will have a look back through the .py file and report back in a bit - just finishing a print and about to have some evening time with the family :) 

it would be nice to try and figure this out before the fix is implemented, but knowing my knowledge limits, I might be better off waiting hahaha! :D

todstelzer, 2023-06-27:
Its line 151 to 153 in the py for bed_xy_position parameter.
And line 158 in my py its set to a fixed value of 175,175. But with bed_xy_position it should work with the original py.

https://github.com/protoloft/klipper_z_calibration/issues/104#issuecomment-1601855604

VORONENTHUSIAST, 2023-06-27:
Yeah, this is where my lack of knowledge is evident - I'm looking at lines 151 through 170 of your .py file and can't understand how it's hard coded to 175, 175? 
[image]

VORONENTHUSIAST, 2023-06-27:
Okay, I'm really confused now - I have downloaded the included .zip from your comment, have looked at the file and then relooked at your screengrab from above - they are vastly different - I will change the file I have here to what your screen grab looks like (obviously switching out 175, 175 to my parameters) and let you know what happens!

todstelzer, 2023-06-27:
That isnt my py file. I redownloaded it and it has the chances in it.

[image]

VORONENTHUSIAST, 2023-06-27:
yeah, would help if I kept up with the avatars next to comments!!

Actually downloaded yours this time! made the change to the section where you put your bed parameters, lets see what happens!!

Thanks for your patience with me! :D

todstelzer, 2023-06-27:
Ah you download the first py file.
You have to use the py file attached to the picture below it

VORONENTHUSIAST, 2023-06-27:
Well, I would test it again, but now for some reason when I unhash my include line for the auto_z_calibrate and the parts inside the print start macro, save the config and reboot, it is ignoring the change and leaving them hashed. I really don't think this is a question for here, as you've done enough for me - so I will get this sorted and then report back if it's working or not. I'm assuming that it will now I have the correct py file!! ;)

VORONENTHUSIAST, 2023-06-27:
Right, everything has saved properly - opened winSCP again to look at the .py, it's the correct one - did the CALIBRATE_Z function - still giving me an out of range error - "!! Move out of range: 381.500 620.000 3.685 [0.000]" 

This is super perplexing!! 

@TitusLabs please comment on here when you have implemented the updates and merged this, or whatever it is you will be doing, looks like I need the professional touch over my hack and slash attempt! :D

todstelzer, 2023-06-27:
Maybe it´s a missing G90.
Try G90 in console and than CALIBRATE_Z

VORONENTHUSIAST, 2023-06-27:
> Maybe it´s a missing G90. Try G90 in console and than CALIBRATE_Z

Can confirm this hasn't worked either :/ something just isn't playing right

todstelzer, 2023-06-27:
Try out my z_calib macro:

> 
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BED_MESH_CALIBRATE_BASE
; gcode parameters
variable_parameter_AREA_START : 0,0
variable_parameter_AREA_END : 0,0
; the clearance between print area and probe area 
variable_mesh_area_offset : 5.0
; number of sample per probe point
variable_probe_samples : 3
; minimum probe count
variable_min_probe_count : 3
; scale up the probe count, should be 1.0 ~ < variable_max_probe_count/variable_min_probe_count
variable_probe_count_scale_factor : 0.7
; enable preference index
variable_enable_reference_index : True
gcode:
    _CheckProbe action=query
    Attach_Probe
    {% if params.AREA_START and params.AREA_END %}
        {% set bedMeshConfig = printer["configfile"].config["bed_mesh"] %}
        {% set safe_min_x = bedMeshConfig.mesh_min.split(",")[0]|float %}
        {% set safe_min_y = bedMeshConfig.mesh_min.split(",")[1]|float %}
        {% set safe_max_x = bedMeshConfig.mesh_max.split(",")[0]|float %}
        {% set safe_max_y = bedMeshConfig.mesh_max.split(",")[1]|float %}

        {% set area_min_x = params.AREA_START.split(",")[0]|float %}
	{% set area_min_y = params.AREA_START.split(",")[1]|float %}
	{% set area_max_x = params.AREA_END.split(",")[0]|float %}
	{% set area_max_y = params.AREA_END.split(",")[1]|float %}

	{% if bedMeshConfig.probe_count.split(",")|length == 2 %}
            {% set meshPointX = bedMeshConfig.probe_count.split(",")[0]|int %}
            {% set meshPointY = bedMeshConfig.probe_count.split(",")[1]|int %}
        {% else %}
            {% set meshPointX = bedMeshConfig.probe_count.split(",")[0]|int %}
            {% set meshPointY = bedMeshConfig.probe_count.split(",")[0]|int %}
        {% endif %}

	{% set meshMaxPointX = meshPointX %}
	{% set meshMaxPointY = meshPointY %}

        {% if (area_min_x < area_max_x) and (area_min_y < area_max_y) %}
            {% if area_min_x - mesh_area_offset >=  safe_min_x %}
                {% set area_min_x = area_min_x - mesh_area_offset %}
            {% else %}
                {% set area_min_x = safe_min_x %}
            {% endif %}

            {% if area_min_y - mesh_area_offset >=  safe_min_y %}
                {% set area_min_y = area_min_y - mesh_area_offset %}
            {% else %}
                {% set area_min_y = safe_min_y %}
            {% endif %}

            {% if area_max_x + mesh_area_offset <=  safe_max_x %}
                {% set area_max_x = area_max_x + mesh_area_offset %}
            {% else %}
                {% set area_max_x = safe_max_x %}
            {% endif %}

            {% if area_max_y + mesh_area_offset <=  safe_max_y %}
                {% set area_max_y = area_max_y + mesh_area_offset %}
            {% else %}
                {% set area_max_y = safe_max_y %}
            {% endif %}

            {% set meshPointX = (meshPointX * (area_max_x - area_min_x) / (safe_max_x - safe_min_x) * probe_count_scale_factor)|round(0)|int %}
            {% if meshPointX < min_probe_count %}
                {% set meshPointX = min_probe_count %}
            {% endif %}
	    {% if meshPointX > meshMaxPointX %}
                {% set meshPointX = meshMaxPointX %}
            {% endif %}

            {% set meshPointY = (meshPointY * (area_max_y -area_min_y ) / (safe_max_y - safe_min_y) * probe_count_scale_factor )|round(0)|int %}
            {% if meshPointY < min_probe_count %}
                {% set meshPointY = min_probe_count %}
            {% endif %}
	    {% if meshPointY > meshMaxPointY %}
                {% set meshPointY = meshMaxPointY %}
            {% endif %}

            {% set algorithm = "lagrange" %}
            {% if meshPointX >=4 or meshPointY >=4 %}
                {% set algorithm = "bicubic" %}
            {% endif %}

            {% if enable_reference_index %}
                BED_MESH_CALIBRATE_BASE mesh_min={area_min_x},{area_min_y} mesh_max={area_max_x},{area_max_y} probe_count={meshPointX},{meshPointY} samples={probe_samples|int} algorithm={algorithm}
            {% else %}
                BED_MESH_CALIBRATE_BASE mesh_min={area_min_x},{area_min_y} mesh_max={area_max_x},{area_max_y} probe_count={meshPointX},{meshPointY} samples={probe_samples|int} algorithm={algorithm}
            {% endif %}
        {% else %}
            BED_MESH_CALIBRATE_BASE
        {% endif %}
    {% else %}
        BED_MESH_CALIBRATE_BASE
    {% endif %}
    Dock_Probe

>

VORONENTHUSIAST, 2023-06-27:
Thanks, can I put this in below the KAMP section of my macro, replacing my BED_MESH_CALIBRATE section, or shall I just use this config as a file in its own right, replacing my one completely for the minute?

todstelzer, 2023-06-27:
Copy your own macro in printer cfg to a text file to save it and paste my macro in and test it.

VORONENTHUSIAST, 2023-06-27:
Added it in, restarted, and got the following error:

File contains parsing errors: /home/mks/klipper_config/Macros/Bed_Mesh_Calibrate.cfg
[line 18]: 'Attach_Probe\n'
[line 19]: '{% if params.AREA_START and params.AREA_END %}\n'
[line 97]: '{% else %}\n'
[line 99]: '{% endif %}\n'
[line 100]: 'Dock_Probe'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Damn I feel like the dog begging for a bone now, I'm so sorry I'm taking up your time!!

todstelzer, 2023-06-27:
Thats my printer.cfg right now.
[printer.zip](https://github.com/protoloft/klipper_z_calibration/files/11886952/printer.zip)

VORONENTHUSIAST, 2023-06-27:
Thanks, copying it from there worked. 

I've just rebooted the printer again by turning it off at the wall and it's still producing the same error, very peculiar.

At this point I'm just going to leave it alone and wait to see what happens. Thanks so much for all your help though!

todstelzer, 2023-06-27:
You can try and ask in the Voron discord Klipper section for help.
I hope you will find the problem. Good luck!
https://discord.gg/voron

VORONENTHUSIAST, 2023-06-27:
They will undoubtedly (and rightly so) berate me for pressing the update button with no good reason... when will I learn, if it ain't broke don't fix it! :')

Luckily I haven't updated the firmware on my other 2.4, so that one will still play ball! :D 

Honestly I can't thank you enough for you help, I'm sure I will have it working again on this machine when the main repo gets updated with the changes!

TitusLabs, 2023-06-27:
So, first of all, thank you for pointing that out. I had not noticed it yet.

Furthermore, you can still use the relative reference index. No printer has to stand still here :smile: 

I have just uploaded a fix in the development branch that should work with both an older Klipper version and with a current Klipper version with relative reference index or zero reference_position configured.

Feel free to give me feedback

VORONENTHUSIAST, 2023-06-27:
Hi, Titus, thanks so much for the response, sadly the relative reference index has now thrown the printer into halting and has to be removed from the config file :/

I will look to update to the fix in the development branch once I've worked out how to do that (I'm really not very good at this side of things) and let you know how it goes :)

TitusLabs, 2023-06-27:
No, the relative reference index is only marked as deprecated. My printer worked just fine after updating Klipper to the current version. The plugin breaks if you replace the relative reference index with the newer zero reference position. You don't have to to that right now...

VORONENTHUSIAST, 2023-06-27:
That's so strange, mine just wouldn't work?! I'm gonna have to go back and re-look at that, as I didn't have any problems before trying to use this zero reference position!

VORONENTHUSIAST, 2023-06-27:
> No, the relative reference index is only marked as deprecated. My printer worked just fine after updating Klipper to the current version. The plugin breaks if you replace the relative reference index with the newer zero reference position (and there is no bed position configured). You don't have to to that right now...

Annnnd now I'm back in another loop I'd forgotten about earlier - since adding the relative reference index back in and removing the zero reference position, I'm being told my z_calibration is not a valid config section - I'm starting to loose the will over this as I can print without it just fine, hahaha

TitusLabs, 2023-06-27:
Ok, may be too many changes at once. Try the new one from the development branch. That should work in any case now. But I don't give any guarantee for any fancy macros here :zany_face:

VORONENTHUSIAST, 2023-06-27:
yeah, it's my hack and slash that's causing the problem - after placing the new file in using winSCP, it keeps loading blank when I turn on the machine - I need to learn more about linux, SSH, etc. and do this properly I think, I have a feeling it's using 3 or 4 different interfaces to make all of my changes! 

I LOVE this macro, it makes my machine feel more 'professional' if you like, having a full calibration before each print starts is really cool to look at too!! :D

VORONENTHUSIAST, 2023-06-27:
IT'S WORKING!!!!!!! 

SO!! instead of replacing the file again, I opened the current one on my machine, pasted the information from your development branch file (my file was blank, no deleting needed) I then saved this and reloaded it. 

After this, I then SSH'd into my Pi (I say Pi, I'm actually using the MKS SKIPR board so have their equivalent on board the PCB) and used the sudo reboot command. This then rebooted my machine and it all loaded - I just ran the Z_CALIBRATE command on my console and low and behold, it worked!!! :D 

I am so happy to have this back up and running again, at least for the minute - I will now need to ponder as to why it's not working with the zero_reference_position ready for when they get rid of relative reference completely!

TitusLabs, 2023-06-27:
Great :+1: 

You always need to reload Klipper when you change a python file!

Now, you should be able to use the zero_reference_position too...

VORONENTHUSIAST, 2023-06-27:
I'll create a separate file using that, upload it and let you know how it goes!!

VORONENTHUSIAST, 2023-06-27:
I have a slight bit of tweaking to do on my switch offset -this is a little higher than I'd like, but the grin on my face definitely isn't!! Thanks so much @TitusLabs and @todstelzer for your help, it's super appreciated!!

[image]

TitusLabs, 2023-06-28:
It's released :+1:

closed 2023-07-05 by TitusLabs