protoloft/klipper_z_calibration #105: Internal error on command:"CALIBRATE_Z"  Internal error on command:"BASE_CALIBRATE_Z"

BelgarionNL, 2023-06-27:
Quad gantry level works fine.

I start the calibrate_z it starts probing on the Z endstop with and without the probe. Then the moment it touches the bed. klipper crashes.

22:46
Internal error on command:"CALIBRATE_Z"
22:46
Internal error on command:"BASE_CALIBRATE_Z"
22:46
Klipper state: Shutdown

BelgarionNL, 2023-06-27:
`
#####################################################################
# Z Calibration
#####################################################################
[z_calibration]
#   The X and Y coordinates (in mm) for clicking the nozzle on the
#   Z endstop.
# nozzle_xy_position: 177.8,248.5
#   The X and Y coordinates (in mm) for clicking the probe's switch
#   on the Z endstop.
switch_xy_position: 178.5,218.5
#   The X and Y coordinates (in mm) for probing on the print surface
#   (e.g. the center point) These coordinates will be adapted by the
#   probe's X and Y offsets. The default is the relative_reference_index
#   of the configured bed_mesh. It will raise an error if there is no
#   probe_bed site and no bed_mesh with a relative_reference_index
#   configured.
bed_xy_position: 120,120
#   The trigger point offset of the used mag-probe switch.
#   This needs to be found out manually.
#   A smaller value is more away from bed >?
# bed_xy_position: default from relative_reference_index of bed_mesh
#   A X, Y coordinate (e.g. 100,100) where the print surface (e.g. the center
#   point) is probed. These coordinates are adjusted by the
#   probe's X and Y offsets. The default is the relative_reference_index
#   of the configured bed_mesh, if configured. It's possible to change the relative
#   reference index at runtime or use the GCode argument BED_POSITION of CALIBRATE_Z.
# wiggle_xy_offsets: 0,0
#   After probing the nozzle and retracting, move x some distance away and
#   back. Useful to prevent the z endstop pin sticking to the nozzle and
#   being pulled out of the assembly. Can be negative. Defaults to zero to
#   disable it. Can be combined in x and y to move diagonally. Be careful
#   to not move your nozzle out of range!
switch_offset: 0.5
#   The maximum allowed deviation of the calculated offset.
#   If the offset exceeds this value, it will stop!
#   The default is 1.0 mm.
# offset_margins: -1.0,1.0
#   The minimum and maximum margins allowed for the calculated offset.
#   If the offset is outside these values, it will stop!
#   The margin can be defined as "min,max" e.g. "-0.5,1.5" or by just one
#   value e.g. "1.0" which translates to "-1.0,1.0" (which is also the default).
# samples: default from "probe:samples" section
#   The number of times to probe each point. The probed z-values
#   will be averaged. The default is from the probe's configuration.
# samples_tolerance: default from "probe:samples_tolerance" section
#   The maximum Z distance (in mm) that a sample may differ from other
#   samples. The default is from the probe's configuration.
# samples_tolerance_retries: default from "probe:samples_tolerance_retries" section
#   The number of times to retry if a sample is found that exceeds
#   samples_tolerance. The default is from the probe's configuration.
# samples_result: default from "probe:samples_result" section
#   The calculation method when sampling more than once - either
#   "median" or "average". The default is from the probe's configuration.
# safe_z_height: default is 2 * z_offset from the "probe:z_offset" section
#   The absolute z position in mm to move to before moving to the next
#   position. The default is two times the z_offset from the probe's
#   configuration. The minimum safe z height is 3mm.
# position_min: default from "stepper_z:position_min" section.
#   Minimum valid distance (in mm) used for probing move. The
#   default is from the Z rail configuration.
speed: 50
#   The moving speed in X and Y. The default is 50 mm/s.
# lift_speed: default from "probe:lift_speed" section
#   Speed (in mm/s) of the Z axis when lifting the probe between
#   samples and clearance moves. The default is from the probe's
#   configuration.
# probing_speed: default from "stepper_z:homing_speed" section.
#   The fast probing speed (in mm/s) used, when probing_first_fast
#   is enabled. The default is from the Z rail configuration.
# probing_second_speed: default from "stepper_z:second_homing_speed" section.
#   The slower speed (in mm/s) for probing the recorded samples.
#   The default is second_homing_speed of the Z rail configuration.
# probing_retract_dist: default from "stepper_z:homing_retract_dist" section.
#   Distance to retract (in mm) before probing the next sample.
#   The default is homing_retract_dist from the Z rail configuration.
probing_first_fast: true
#   If true, the first probing will be faster by the probing speed.
#   This is to get down faster and not record the result as a
#   probing sample. The default is false.
start_gcode: _SET_Z_CURRENT VAL=HOME
             DETACH_PROBE
#   A list of G-Code commands to run before each calibration command.
#   See docs/Command_Templates.md for the G-Code format. This can be used to
#   attach the probe.
before_switch_gcode: ATTACH_PROBE
                     G0 Z20
#   A list of G-Code commands to run before to each probing on the
#   mag-probe. See docs/Command_Templates.md for the G-Code format. This can
#   be used to attach the probe after probing on the nozzle and before probing
#   on the mag-probe.
end_gcode: _SET_Z_CURRENT
           DETACH_PROBE
#   A list of G-Code commands to run after each calibration command.
#   See docs/Command_Templates.md for the G-Code format. This can be used to
#   detach the probe afterwards.
`

BelgarionNL, 2023-06-27:
> 
#####################################################################
# Z Calibration
#####################################################################
[z_calibration]
#   The X and Y coordinates (in mm) for clicking the nozzle on the
#   Z endstop.
# nozzle_xy_position: 177.8,248.5
#   The X and Y coordinates (in mm) for clicking the probe's switch
#   on the Z endstop.
switch_xy_position: 178.5,218.5
#   The X and Y coordinates (in mm) for probing on the print surface
#   (e.g. the center point) These coordinates will be adapted by the
#   probe's X and Y offsets. The default is the relative_reference_index
#   of the configured bed_mesh. It will raise an error if there is no
#   probe_bed site and no bed_mesh with a relative_reference_index
#   configured.
bed_xy_position: 120,120
#   The trigger point offset of the used mag-probe switch.
#   This needs to be found out manually.
#   A smaller value is more away from bed >?
# bed_xy_position: default from relative_reference_index of bed_mesh
#   A X, Y coordinate (e.g. 100,100) where the print surface (e.g. the center
#   point) is probed. These coordinates are adjusted by the
#   probe's X and Y offsets. The default is the relative_reference_index
#   of the configured bed_mesh, if configured. It's possible to change the relative
#   reference index at runtime or use the GCode argument BED_POSITION of CALIBRATE_Z.
# wiggle_xy_offsets: 0,0
#   After probing the nozzle and retracting, move x some distance away and
#   back. Useful to prevent the z endstop pin sticking to the nozzle and
#   being pulled out of the assembly. Can be negative. Defaults to zero to
#   disable it. Can be combined in x and y to move diagonally. Be careful
#   to not move your nozzle out of range!
switch_offset: 0.5
#   The maximum allowed deviation of the calculated offset.
#   If the offset exceeds this value, it will stop!
#   The default is 1.0 mm.
# offset_margins: -1.0,1.0
#   The minimum and maximum margins allowed for the calculated offset.
#   If the offset is outside these values, it will stop!
#   The margin can be defined as "min,max" e.g. "-0.5,1.5" or by just one
#   value e.g. "1.0" which translates to "-1.0,1.0" (which is also the default).
# samples: default from "probe:samples" section
#   The number of times to probe each point. The probed z-values
#   will be averaged. The default is from the probe's configuration.
# samples_tolerance: default from "probe:samples_tolerance" section
#   The maximum Z distance (in mm) that a sample may differ from other
#   samples. The default is from the probe's configuration.
# samples_tolerance_retries: default from "probe:samples_tolerance_retries" section
#   The number of times to retry if a sample is found that exceeds
#   samples_tolerance. The default is from the probe's configuration.
# samples_result: default from "probe:samples_result" section
#   The calculation method when sampling more than once - either
#   "median" or "average". The default is from the probe's configuration.
# safe_z_height: default is 2 * z_offset from the "probe:z_offset" section
#   The absolute z position in mm to move to before moving to the next
#   position. The default is two times the z_offset from the probe's
#   configuration. The minimum safe z height is 3mm.
# position_min: default from "stepper_z:position_min" section.
#   Minimum valid distance (in mm) used for probing move. The
#   default is from the Z rail configuration.
speed: 50
#   The moving speed in X and Y. The default is 50 mm/s.
# lift_speed: default from "probe:lift_speed" section
#   Speed (in mm/s) of the Z axis when lifting the probe between
#   samples and clearance moves. The default is from the probe's
#   configuration.
# probing_speed: default from "stepper_z:homing_speed" section.
#   The fast probing speed (in mm/s) used, when probing_first_fast
#   is enabled. The default is from the Z rail configuration.
# probing_second_speed: default from "stepper_z:second_homing_speed" section.
#   The slower speed (in mm/s) for probing the recorded samples.
#   The default is second_homing_speed of the Z rail configuration.
# probing_retract_dist: default from "stepper_z:homing_retract_dist" section.
#   Distance to retract (in mm) before probing the next sample.
#   The default is homing_retract_dist from the Z rail configuration.
probing_first_fast: true
#   If true, the first probing will be faster by the probing speed.
#   This is to get down faster and not record the result as a
#   probing sample. The default is false.
start_gcode: _SET_Z_CURRENT VAL=HOME
             DETACH_PROBE
#   A list of G-Code commands to run before each calibration command.
#   See docs/Command_Templates.md for the G-Code format. This can be used to
#   attach the probe.
before_switch_gcode: ATTACH_PROBE
                     G0 Z20
#   A list of G-Code commands to run before to each probing on the
#   mag-probe. See docs/Command_Templates.md for the G-Code format. This can
#   be used to attach the probe after probing on the nozzle and before probing
#   on the mag-probe.
end_gcode: _SET_Z_CURRENT
           DETACH_PROBE
#   A list of G-Code commands to run after each calibration command.
#   See docs/Command_Templates.md for the G-Code format. This can be used to
#   detach the probe afterwards.

BelgarionNL, 2023-06-27:
[config-20230602-225305.zip](https://github.com/protoloft/klipper_z_calibration/files/11887180/config-20230602-225305.zip)

TitusLabs, 2023-06-27:
Is this the only error in your log? And this error is repeatable?

BelgarionNL, 2023-06-27:
Exception in flush_handler
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

yep every single time. the moment the clicky probe touches the bed for z calibration its dead.

reinstalled klipper on new sd card + updated the skr boards with the latest klipper.

BelgarionNL, 2023-06-27:
[klippy (5).log](https://github.com/protoloft/klipper_z_calibration/files/11887736/klippy.5.log)
[config-20230602-225305.zip](https://github.com/protoloft/klipper_z_calibration/files/11887737/config-20230602-225305.zip)

BelgarionNL, 2023-06-27:
I thought it had something to do that I dont probe my bed because its glass so its always flat.

TitusLabs, 2023-06-27:
You also had this error in your dockable probe extension. I don't think this is related to z calibration.

I found this:
```
  File "/home/pi/klipper/klippy/stepper.py", line 228, in generate_steps
    raise error("Internal error in stepcompress")
stepper.error: Internal error in stepcompress
```
and this error:
```
  File "/home/pi/klipper/klippy/stepper.py", line 180, in note_homing_end
    raise error("Internal error in stepcompress")
stepper.error: Internal error in stepcompress
```

Here is a first starting point on this: https://github.com/Klipper3d/klipper/issues/4097

BelgarionNL, 2023-06-27:
I cleared the klipper log and ran calibrate_z once. 
[klippy (6).log](https://github.com/protoloft/klipper_z_calibration/files/11887991/klippy.6.log)

BelgarionNL, 2023-06-27:
you were correct. I tried the new dockable_probe (https://klipper.discourse.group/t/reviving-the-dockable-probe/4256/51) instead of the one I had been using for years from mental.

reverted back to old one and it works fine.

kdb424, 2023-06-27:
Confirmed this is the case, and mentioned in their PR. Mental's module works fine so this is an external issue.

TitusLabs, 2024-06-16:
Is this still a problem? I had unfortunately forgotten the PR regarding the trial session on the develop branch. However, it is now released. But I'm not sure if this is somehow related to this :thinking:

closed 2024-08-01 by TitusLabs