protoloft/klipper_z_calibration #106: Bed Mesh adds to my Z calibration

canadank0, 2023-06-28:
After i updated to the latest Klipper the BED_MESH_PROFILE LOAD in START_PRINT macro adds value to Z calibration number so my nozzle crashing into the bed. When you calibrate the Z? before the BED_MESH_PROFILE LOAD or after? How do you subtract the values? My bed shape is kind of terrible so i really need the  bed mesh to keep consistent first layer.

TitusLabs, 2023-06-28:
If setup correctly and you do not use adaptive bed mesh (which you don't because you want to load a mesh), the order should make no difference. But, you can calibrate z after loading/creating the mesh. However, it is important that you set up a relative_reference_index or the newer zero_reference_position under mesh (but for the last one, you currently need the development branch of the z calibration). The z calibration must probe the bed at this position which is done automatically if you not configure the bed_xy_position in z_calibration.

JarTheHead, 2023-06-28:
I believe I'm currently on the development branch, however it still wants me to set a bed_xy_position with my adaptive mesh (using KAMP), even though the zero_reference_position does show up using BED_MESH_OUTPUT PGP=1. My personal error still says relative_reference_index, though so is that a sign it didn't update? If not, is there a specific way I need to setup KAMP to use a zero_reference_index?

TitusLabs, 2023-06-29:
It's released now with support for the zero_reference_position :+1: 

I cannot say anything about KAMP because I don't use it. But, actually everything should remain the same. I'm not sure if KAMP uses the bed position parameter or changes the relative_reference_index. If it's the last one, the macro has to be changed....

canadank0, 2023-06-29:
cool, thanks

closed 2023-06-29 by canadank0